builder555/PineSAM #2: could this run on windows or does it rely on linux for the bluetooth

TomW1605, 2023-02-06:
basicaly all in the title

builder555, 2023-02-06:
I don't have a windows machine to test it, and my VM is notoriously picky about bluetooth.

TomW1605, 2023-02-06:
ok i cant get it working on windows (also had issues that there is no official pyenv for windows), just wont detect the pinecil even though its paired and connected so im guessing the BLE stuff only works on linux and while i have plenty of linux machins around none of them have bluetooth

builder555, 2023-02-06:
Ideally, you shouldn't have to pair anything. You turn on your iron, start the script and it should work. I'm using [bleak](https://pypi.org/project/bleak/) library, and it says it's compatible with Win10. Do you see any errors in terminal?

TomW1605, 2023-02-06:
nothing in the python console, in the web console it is just a device not found error, `{status: 'ERROR', message: 'Device not found', command: 'GET_SETTINGS'}`

TomW1605, 2023-02-06:
im wondering if it is the address here https://github.com/builder555/pinecil-v2/blob/master/pinecil_ble.py#L76

is that the same for every pinecil? i know the BLE MAC address isnt

builder555, 2023-02-06:
yup, i just noticed it, removed.

TomW1605, 2023-02-06:
yep that did it

TomW1605, 2023-02-06:
out of intrest, where did you get that full address? i can get the mac address but cant find any way to get the full one

closed 2023-02-06 by TomW1605

builder555, 2023-02-06:
That was a good old-fashioned "let me hard-code this just for now, i'll remove it before committing". Thanks for reporting it!

On MacOS it shows the UUID instead of the Mac address: https://bleak.readthedocs.io/en/latest/api/index.html?highlight=uuid

TomW1605, 2023-02-06:
ah right that makes sense, so im guessing that if i put the mac address for mine in it would have worked too but i didnt think to try because i assumed it was a diffrent address

River-b, 2023-02-06:
> is that the same for every pinecil? i know the BLE MAC address isnt

the BLE name was  "pinecil"  for all of them  but Ralim asked me to put an Issue  Ticket into  main IronOS  a couple weeks ago for it to be "Pinecil+unique number". it probably will be changed in future  firmware as it's already being worked on. 
https://github.com/Ralim/IronOS/issues/1545

River-b, 2023-02-06:
are there  instructions  for how to set this  BLE settings for  Windows people?  Although I have access to  Mac,  I'm mostly on Windows testing the BLE features since  Joric's  Pinecil BLE browser pages  runs best on Windows PC and Android phones too. 
if  I could get link to instructions that would be idea, then I could add link to Pinecil  Wiki for people to come to, or  work with someone to make instructions more clear and usable for larger number of people who are not familiar with Github.

builder555, 2023-02-06:
@River-b i'll try to get a hold of a windows image to run on my machine and write down the steps tonight, haven't had the best luck with bluetooth on VMs.

River-b, 2023-02-06:
this works  on any browser?  could it work on Android Phones too so people could have alternate display on phone and change set  temperature?

builder555, 2023-02-06:
@River-b yes, but the backend has to run on a computer, then you can access it from any phone/tablet (including iphones) on the same network. The only caveat is that the iron needs to be within the bluetooth range of the computer running the backend. I'll update the repo in a couple of hours with those changes.

River-b, 2023-02-07:
> please try it out now

suddenly got busy with work. will give feedback when I can get to this later this week. thanks