builder555/PineSAM #3: Set Defaults button

River-b, 2023-02-06:
Is there a way to have a button that will "Set All to Defaults"  which will restore everything to exactly the way  it is released with the  firmware version the person is on.  Ralim  sends out all firmware with a specific set of defaults that he thinks more people will use.  
sometimes people change so many settings and don't know what they messed up , they just want an easy way to restore it all quickly  and get all default sleep, hall sensor, temperature default settings.  We have this ability inside Advanced menu now.

builder555, 2023-02-06:
I'll check if it's available through BLE interface.

builder555, 2023-02-07:
@River-b doesn't look like it's exposed, at least I can't find it. You can create an issue with [IronOS](/Ralim/IronOS).

River-b, 2023-02-07:
> @River-b doesn't look like it's exposed, at least I can't find it. You can create an issue with [IronOS](/Ralim/IronOS).

https://github.com/Ralim/IronOS/issues/1566

builder555, 2023-02-11:
branch [v2.21](../tree/v2.21) works with [this](https://github.com/Ralim/IronOS/suites/10919790100/artifacts/551993774) artifact. I'll wait before merging in case @Ralim makes other changes. It's also backwards compatible

PR #6 (2023-02-12): compatible with v2.21, adds reset factory settings. closes #3
Works with Ralim/IronOS#1567
closes #3 (Requires artifacts from ^)
commit 4f565990 (2023-02-12): Merge pull request #6 from builder555/v2.21

compatible with v2.21, adds reset factory settings. closes #3
closed 2023-02-12 by builder555