builder555/PineSAM #8: windows 11 install - getting errors

River-b, 2023-02-15:
I could use a hint for how to install this on windows 11.

1. updated newest python for windows today and checked off  add to enviromment variables in the Custom install.
2. using Windows Powershell as administrator.  

getting errors

[image]

builder555, 2023-02-15:
What happens if you run it in regular terminal instead of powershell?
Also, can you run this command?

`python -m pip --version`

River-b, 2023-02-15:
> python -m pip --version

for some reason windows still  is pointing to older 3.6 python. 
[image]

I need to see how to uninstall that. I just updated to 3.11 so don't know why it's not using that.

builder555, 2023-02-15:
i saw this post: https://stackoverflow.com/a/53579844/1913115 seems like a similar issue

River-b, 2023-02-15:
uninstalled all old versions of python from windows add/remove.
Reinstalling 3.11 fresh now.  should I check the box atthe bottom
"Add python.exe to PATH"
[image]

builder555, 2023-02-15:
yes

River-b, 2023-02-15:
start.bat is now working in Windows Powershell and it lauched a Chrome browser window with the app.
Perhaps you could make a name for this app so it's easier for me to refer to it in the Discord and other chat rooms on pinecil :)

For reference this is what people with Windows Powershell will see when the app successfully launches, and see a separate browser window open. I like to use Powershell over old Command as Powershell has more autocomplete features and works better for everything.

[image]

builder555, 2023-02-15:
I might play around later with Pyinstaller to see if i can get GH actions to build binaries instead of putting you and others through this nightmare again 😅 

As for the name, I hadn't thought of it. I built this because I am lazy and wanted to see all the settings with big buttons at once. Perhaps `Pinecett`? some sort of pinecil+settings amalgam

River-b, 2023-02-15:
After that initial window since Python and Pip were giving me messages, the next time I ran the script it was much simpler output.

Below is more normal for windows users after the initial install (since I had bunch of unrelated notices to update).
[image]

River-b, 2023-02-15:
> nightmare again

it was ok once I uninstalled all 4 versions of python I did not know I had. Python installed does not give you option to remove older versions and the path is set to original.  Another tip I learned from stackexchange is when installing select  checkbox "install for all users". they said if you select check box install for all users, then you don't have to "Add python.exe to path" on first screen. but I did both options.

maybe add that to windows install instructions:

1.  Select Custom Installation
2. At the bottom, both boxes should be checked, use admin privileges  and Add python.exe to PATH
3. on follow screens, check to box to "Install Python for all users" (this also sets the PATH so technically if install for all users is selected then "Add python.exe to PATH" is not needed, but doing Both is  fine and is what I did.

River-b, 2023-02-15:
> As for the name, I hadn't thought of it. I built this because I am lazy and wanted to see all the settings with big buttons at once. Perhaps `Pinecett`? some sort of pinecil+settings amalgam

PineSetter
PineSet-BLE

Another thing that would make it more user friendly  is have the Temperature Slider as the main thing.
When I use Joric's BLE Pinecil webpage, I am mostly just using my Android phone, sliding the temperature up/down as needed to actually use Pinecil without doing it on Pinecil Screen. Android is nicer color display and bigger. I keep Phone in a stand while soldering.
Ideal would be something like way Joric has main notices  like Temperature display with slider for quick changing temperature bigger, and also shows actively the volts/watts  you are using.   
then have a separate Gear icon or  3 dots like android uses to see "more settings "  and then inside there are all the other user  settings that get flashed.  that is where the warning message is more important and why people should do all of their setting changes  at same time and then do one commit to update the flash. 

For everyday usage then I would mainly slide temperature up/down and see the power  actively, but then if I click Gear or 3-dots I see all the other settings possible to change.

this is the Joric Github I was testing before https://joric.github.io/pinecil/
(https://github.com/joric/pinecil)
It is good for everyday use if you have a laptop or Android phone while soldering and can just use finger on phone screen to quickly change temperature. but does not give access to change settings.

River-b, 2023-02-15:
closing ticket as it is working now

closed 2023-02-15 by River-b