builder555/PineSAM #11: Show the Unique name that Pinecil is broadcasting at the top of the App

River-b, 2023-02-17:
Show the Unique name that Pinecil is broadcasting at the top of the App page somewhere or something I can click on to see it?

- I'm trying to test the new IronOS beta from here https://github.com/Ralim/IronOS/actions/runs/4151093715
- Ralim implemented Uniue BLE name for broadcasting Pinecil-XXXXX 
- See Ralim's notes on this here: https://github.com/Ralim/IronOS/pull/1567

using Joric's webpage, both my V2's show up as the exact same  Pinecil-5xxxx  (5 digit number)
even if I go to different devices like android phone or laptop. Another tester today said he only got 4 numbers and same issue, same number showed up on multiple reboots and connections, his was Pinecil-3xxx. it's different number from me, but it's not randomly changing.

is it possible on your App for me to see what the Unique name is that  pinecil is broadcasting to confirm this? some people like me have 3x pinecil V2 ( 1/3 is currently needing repair) so unique name that Ralim is trying to implement is good so we can tell them apart.  
I'm in the Pinecil Settings APP now but I don't see anything displayed that looks like Pinecil-5####

Because I am in unique position of owning more than one V2, I am wanting to double check if what I see in Joric's pairing  (same broadcasted name  Pinecil-number on 2 separate pinecils)  is also showing in your APP.

builder555, 2023-02-17:
> Show the Unique name that Pinecil is broadcasting at the top of the App page somewhere or something I can click on to see it?

Not at the moment, but let me work on that.

> Because I am in unique position of owning more than one V2, I am wanting to double check if what I see in Joric's pairing (same broadcasted name Pinecil-number on 2 separate pinecils) is also showing in your APP.

You lucky duck, you!

builder555, 2023-02-18:
so this is strange, i've tried probably 4 or 5 different artifacts from different builds, including the one you linked here, but all of them just show up as "Pinecil" in BLE scanning (including Joric's UI)

River-b, 2023-02-18:
2 of us tested this one today - the newest beta https://github.com/Ralim/IronOS/actions/runs/4190920103

and I also additional tested this one
https://github.com/Ralim/IronOS/actions/runs/4151093715

Both of them give me the name  Pinecil-5xxxx on Joric's  BLE page. and also zombie_detox got a number but they got 4 digits and I got 5 digits. 

I also Posted the details of what we did for Ralim to see so we could get clarification on how he is creating the random number here:  https://github.com/Ralim/IronOS/pull/1567

in stable 2.20 it simply showed as  "Pinecil" in Joric's app.
with the 2 new beta's I tested  and  Zombie_detox also.   we get a number now similar to how Ralim discribes in #1567.

With the two beta's I linked, I see this in Joric's BLE now. it's blanked out but it's  5 numbers.
Pinecil-5xxxx   basically is my number. 
[image]

The strange thing is I am seeing this same exact number no matter which of my two Pinecils I try and they have completely different ID numbers, serial numbers and manufacture dates. got one in August and one 1-2 months later. This is why I asked Ralim to clarify how the "random" number is being made since it's not working it seems in the testing some members have done so far.

However, both of us do get a number now.  Pinecil-xxxx  and Pinecil-xxxxx  
I also think it's odd I get 5 digits which start with a "5"  and zombie got 4 digits, and starts with a "3".

River-b, 2023-02-18:
If you are not even getting a number and you are using the same beta, this is yet another issue with this BLE beta update. maybe let Ralim know.  something is amiss with the BLE Unique name /random number code?

River-b, 2023-02-18:
besides the 3 separate machines all giving me exact same Pinecil-number I also tried Vivaldi browser to joric Ble  and it was the same exact number.

River-b, 2023-02-18:
Builder, Ralim just released a fix  for the Unique names about 10 min ago.

Try this one  it might work for you now.
**https://github.com/Ralim/IronOS/actions/runs/4210122189**

Basically  2 of my pinecils  and also Tom's pinecil from across planet all share same exact last 8 characters for Serial number (the number under the ID in debug menu)  so we all 3  got same exact Pinecil-55236   "unique" name (not).
but with this new code just now.   Tom and my 2 pinecil's are all 3 now  slightly different and Unique names.   instead of  all numbers, there are letters now too 

This is how ralim changed code just now to fix it as he mentioned in the pinecil-dev chat.
[image]

builder555, 2023-02-19:
@River-b do your two pencils show different IDs with the new PineSAM?

River-b, 2023-02-19:
> @River-b do your two pencils show different IDs with the new PineSAM?

Yes I just tested new beta IronOS  with newest PineSam linked in readme page.  Both  show  Unique name in upper right of  browser.   however as you can see in the other ticket  I am now having issues connecting one of the two  Pinecils to  the app.

I'm not sure why but it's taking me 2-4 times running the script before  2nd pinecil will connect.

closing this ticket as resolved since  I can  see Unique  name in PineSAM now

closed 2023-02-19 by River-b