builder555/PineSAM #12: Allow Solder Temp & Boost Temp an option to type exact number directly into the box for changing temperature

River-b, 2023-02-17:
sorry for submitting all of these at once. I just had some time today to do more testing with the app.

**Is your feature request related to a problem? Please describe.**
Sometimes it's hard for my mouse hand to easily slide the bar to change the temperature setting here accurately, i.e. I want to set it to 306 C. I suspect this would also be same if I got this working on Android with touch screen too.

**Describe the solution you'd like**
 Is it possible for the number boxes on Soldering temp and Boost Temp to allow people choice to type the exact number desired into the box instead of using the slider (both number entry or slider as a choice)?

[image]

**Additional context**
I still like slider too, just sometimes mouse is frustrating and would be good to just type it in. I think on Android phone I would more likely use the slider but on PC with keyboard, typing a number directly into the box is faster/more accurate.

builder555, 2023-02-17:
no worries. it'll give me something to work on during the weekend :)

builder555, 2023-02-19:
You should be able to click the value and change it directly. The value will be saved once you navigate out of the box.
the commit automatically closed the issue, but feel free to re-open it if something else is missing @River-b

River-b, 2023-02-19:
new download does not seem to be working for both my Pinecils.
Both pinecils work on Joric's BLE. Both have same beta  dated feb 18, 2023 (more unique).

One of my Pinecils works on PineSam ( I downloaded fresh zip from main page 20 min ago to test temperature change feature).

Second V2, does not and gets these errors
[image]
[image]

River-b, 2023-02-19:
nvm  after doing it a few more times it worked. maybe there is a minimum amount of time we need to give Pinecils  to be on before starting the app so they have time to properly be seen by PC.

Strange thing is  the working Pinecil, I literally  just plugged it in started the script and it worked right away.  I didn't really wait or delay.

River-b, 2023-02-19:
Testing the new temp Box.
I prefer it does not ask for a confirmation every time.
I like that the slider you just slide the bar and let go and it would just do it as soon as you let go of mouse sliding it.
With the new temperature Box  after I hit Enter  it makes me do a separate confirmation step.

[image]
and this is what it looks like on script 
[image]

**Outcome desired:**

1.  have it like the slider bar where no additional confirmation is needed.  
2. user clicks on box, enters a number, then hits  `enter` . Then enter alone should be confirmation.
3. White boxes in User interface menu are easier to see. maybe make temp change boxes also white. and the size of boxes in user interface is better too. 
4. maybe change everything to left justified?  the different items are  shifted different spots so it's harder to follow them down the page.
[image]
5. Make the slider bar just a little longer on soldering temp and boost.
6. suggestion to remove the green background and just have thick colored boxes around section to make it less busy.

River-b, 2023-02-19:
when I enter 455C  which is out of range I get this warning but it still sets the number in the box
[image]
but leaves the pinecil unchanged.   I suggest if someone enters too high a number past the limit of 450C,  give the "value out of Range"  error message but also put number in box back to  what they had and what is seen on Pinecil Screen.  
if they had 290C before, and they enter  455C,  pop up error message and put box back at 290C  since that is what is Still on the pinecil in real life since it was unchanged.  Don't put 455C in the box

River-b, 2023-02-19:
> but feel free to re-open it if something else is missing

I don't know how to reopen the ticket I dont seem to have a button for that :) but you can see all my testing and comments

River-b, 2023-02-19:
I just did test all over again to see what is going on. I'm using Windows 11, Chrome browser, and windows Powershell for terminal.

Good pinecil = P-a.  Bad pinecil = P-b

- P-a  plugged in, ran script . worked Right away fast and Browser window opened with no issue.

- P-b  same issue.  I did not close my browser but I close all tabs with  PineSam open.  and I can't connect 
This is what script shows on  P-b
[image]

Do you think there is something happening for people with two or more pinecils?

River-b, 2023-02-19:
Problem is  resolved with P-B  bad pinecil. 
Rebooted whole windows 11 PC. 
Now both P-b and P-a  pinecils  connect with Script on first attempt. 

Don't know why but  reboot fixed everything.  Suggest  this is suggestion with users who report  connection issue in script or need to try 3-4 times to connect. simple reboot of whole windows PC is solution.

builder555, 2023-02-20:
wow that's a lot of stuff :)

ok i see what's happening with the "are you sure" box - i forgot to disable the "submit on enter" on the page and "submit" makes a call to "reset all settings". whoops!
anyway, that's fixed now in #45756ce - it'll save when you hit Enter properly without pop-ups.

I'll work on resetting the value back, if there's an error.

as for two pinecils... i wouldn't be able to troubleshoot it as I only have one. theoretically it's possible to build a custom firmware for it to advertise it as a "different" device every time, but I just don't have that kind of time, anyway, if after a reboot it works, maybe we can leave it at that for now.

as for colour and UI tweaks, can you create a separate issue for that? you can just edit your comment and cut/paste it into a new one

River-b, 2023-02-20:
yes i'll make a new ticket about colors/format

builder555, 2023-02-20:
let me know if entering the values works as expected now or needs further tweaking

River-b, 2023-02-20:
1. Tested out of range **Set temperature** - works and completed. typing 455  or 5  °C pops up helpful error message and leaves the Set temperature the same as the current one.
2. Test Boost Temperature - also works, can not enter anything not in the range of 250-450 °C. 
3. Tested °F also works at the adjusted F numbers.
[image]

commit 7127d992 (2023-02-19): able to click on the range value to change it directly

hopefully closes #12
closed 2023-02-20 by River-b