builder555/PineSAM #14: Add version number, and Icon with  link to Github PineSAM

River-b, 2023-02-19:
1. Suggestion to have a version number for PineSAM  that shows in tiny font at top left and make "Pinecil V2 Settings and Menu" a lot smaller font and possibly just move it down to next line

### PineSAM     (PineSAM  can be same bold)
v 1.00 13-02-2023   (version number/date in very small, non bold font)

[image]

2. PineSAM - Pinecil V2 Settings and Menus  makes top  a little busy.   just PineSAM alone by itself would be fine :)  don't need the "Pinecil V2 Settings And Menus " as people can get that inside the readme.  keeps app cleaner less clutter to distract from functions .  however having a version number is good for communication so you know if people are using the newest app or an old one.
3. Could add a Github cat  to the corner like in Joric's BLE.  that github cat  is animated and if you click on it , it opens link to the Github Dev.   Would be very nice and convenient to have same on PineSAM  so  users can get back to main repo fast to check for updates and  read notes also if there is an issue.  
I don't know where Joric got the Github Cat icon, but when you hover over it,  the cat waves at you to indicate it does  something and when you click it, it opens the Dev on Github which is  useful to open issue ticket too. 
This also makes screen  clean and not cluttered  with long link name.  
[image]

River-b, 2023-02-19:
In the Readme page also. I'm not sure when I click on Windows download which zip it is and if it's a new version or same old one.  only way to know is to try it and see if I see changes. 

if there is a way to name the link with version number or something so that people know it's a newer one than they downloaded last week or something.   this is minor but good to know for future :)

builder555, 2023-02-20:
good suggestions about the font/version tweaks, i can work on that.
i can probably just copy the cat from Joric's page, unless he paid for it.

the link to the zip file is always the latest, it's the same as clicking CODE > DOWNLOAD 

the version thing is a bit more tricky to get right. i was actually looking into it already, just need to set it up to update automatically without having to remember, otherwise there'll be even more confusion.

River-b, 2023-02-20:
https://tholman.com/github-corners/

builder555, 2023-02-20:
ahhh, looks like adding automatic version numbers also created releases. can't be bothered today to figure out how to hide those, let's see if people start complaining 🤷

River-b, 2023-02-20:
> ahhh, looks like adding automatic version numbers also created releases. can't be bothered today to figure out how to hide those, let's see if people start complaining 🤷

you have a Release Zip now :)     Grats on your first release.   you think that will lead to complaints?

River-b, 2023-02-20:
is it possible to  Zip the README   with the Zip  so there is an instruction text with it? that might help

builder555, 2023-02-20:
>you mean complaints because they grab the release zip but then don't know what to do because they skip Readme?
yup

> is it possible to Zip the README with the Zip so there is an instruction text with it? that might help

it's already there. hopefully people read it :)

River-b, 2023-02-20:
Closing ticket as completed.
version number works and Github cat works to redirect to PineSAM project.
[image]

closed 2023-02-20 by River-b