builder555/PineSAM #15: Format changes, left Justified. UI tweaks

River-b, 2023-02-20:
All of this is optional of course and only if you have time and feel like it :)

1.  White boxes in User interface menu are easier to see. maybe make temp change boxes also white.  the size of boxes in user interface is better too. 
2. maybe change everything to left justified?  the different items are  shifted different spots so it's harder to follow them down the page.
3. Left justified so the eye doesn't have to work  hard. 
[image]

4. once everythign is left justified, make the slider bar just a little longer on soldering temp and boost.
5. suggestion to remove the green background and just have thick colored Box sections around Category of grouped settings to make it less busy. 
6. for example Power Settings would we  one box all white with  color main border.  and the  main  heading Power Settings with drop down arrow would be only thing with full background color. 

[image]
7. maybe each  sub-menu item like PD timeout  don't show the scrolling help message. maybe user has to hover over word PD Timeout to see pop-up with help text.  OR    on left side of "PD timeout"  have a small flyout arrow and if you click it it opens to show help message. that would clean this up a lot. because people don't need to see the help messages all the time.  they get used to what the items does. and if they forget they can click to see help message or click arrow for flyout?   just a suggestion. 
8. Collor of the  sliding bards should be consistent.  with color of the box info next to it  so whatever  border color is used for temperature and boost boxes for example, make the slider  bar the same color.  

These are all just thoughts, I'm not sure it will look good or worse with my suggestions. I do think all left justified will help and makeing all  cartegories color like  Power Settings, Sleep mode  but having the main boxes be all white.  with a color border around everything. More white space to keep things less clutterered. 
Could go with Teal green/Orange/gray theme. or  a Purple and teal theme  like image by sticking to just 3 colors and grey and black for stuff but mostly white everywhere.
[image]

If you want some color background  for the boxes.  could go with a very light color like this entry boxes are White but see that the rest of the page is a very light share of bone white.  and there is even a light grey shade there. 
[image]

River-b, 2023-02-20:
[image]

River-b, 2023-02-20:
[image]

builder555, 2023-02-20:
I'll play around with the colours and the alignment, see what comes of it. We'll need to think about the descriptions though. Generally hover is great for the "clean look", but it's not recommended for the accessibility. Not that the app is terribly accessible right now, but this will make it even more difficult. Perhaps some kind of switch or a question mark that reveals all the descriptions? Not sure.

River-b, 2023-02-20:
> not recommended for the accessibility.

agreed, on a Phone clicking an arrow next to name of function which shows the help message might be better. then click arrow again to collapse.
you have arrows now which collapse sections and is  handy

builder555, 2023-02-21:
[image]
what about something like this? @River-b

builder555, 2023-02-21:
and this is with dark reader
[image]

builder555, 2023-02-21:
[this](https://github.com/builder555/PineSAM/archive/refs/heads/colour-tweaks.zip) will be the latest colour/ui tweaks branch that you can download

River-b, 2023-02-21:
download this one and not from Readme?

builder555, 2023-02-21:
correct

River-b, 2023-02-21:
will this have the iphone fix?

builder555, 2023-02-21:
v1.0.13 should already have the iphone fix. you might need to refresh the page on iphone, sometimes it loads cached stuff

River-b, 2023-02-21:
I see this on PC windows chrome 
[image]

builder555, 2023-02-21:
it requires a few refreshes to get the css file loaded, i'll need to add cache-busting feature

River-b, 2023-02-21:
can i do something on my end like clear browser data?

builder555, 2023-02-21:
ctrl+shift+i, then you should be able to right-click on the "refresh" button on your browser (not the one inside PineSAM) and select "Empty Cache and Hard Reload"

River-b, 2023-02-21:
1. I see the new display in windows and iphone - works
2. left justified is a lot better :)
3. Show hints button is a nice simple solution.
4. this is much cleaner and easy to work with, especially on a phone. 
5. cat could be pink or light purple, or something that compliments and doesn't distract from the color scheme.
6. Unique name is a little large, it's not something people need to refer to much except for troubleshooting. make it a smaller font.  maybe put  "Paired"  `Pinecil-D1234567`  as a grey unchangeable box  with the name.  grey like the version number, but a little bigger than version number font.
7. where you have the Pop-out drop down, I like that it gets bigger and easy to see and select it on the iphone and browser like "start-up behavior".
8. Blue boxes need  to be a little bigger if this is not too hard. this is good for people who have to wear glasses. could be bigger on both browser and phone versions.
9. on the phone so far I like using the sliders horizontally.
10. app  on phone does not let me turn horiztonal display 
11. if you are doing a "main"  work screen then  can ignore for now that the Set Temp or Set °C is too small for good user interaction.  if  there will be a main screen with this bigger and you can wear  nitrile gloves that is  good.   
12. Right now  Set Temp area doesn't show if it's C or F.   would be nice to translate whatever someone put in the UI settings  and  in the Top   put  Set °C  for example before the blue box and slider.

River-b, 2023-02-22:
white space on left is good now for all headers but not for Hint help messages
just need a little white space on left for them

[image]

River-b, 2023-02-22:
after white space is fixed I think  this ticket can be closed

commit c21ea391 (2023-02-22): fix: #15
closed 2023-02-22 by builder555