builder555/PineSAM #21: Plus and minus with slider  on temperatures.

River-b, 2023-02-20:
if I'm on a phone versus a PC using this. I would probably want a  `[-]` and a` [+] ` on either side of the  Slide bar for temperature changes.  this way I could just tap with my finger  if sliding was not accurate enough.  since on a phone it's not as easy to type a number and one might just want to touch tap a [-] or a [+]  button which is on either side of the slide bar.

on the main Temperature  or  Set temp  only.  since that is the one we change the most often. 

Question -  would the tapping follow the increments set in the Temperature Increments - short and long press button?   for example if I  selected short press to be increments of "5"   then  tapping [-]  or [+]  would increment in  5 degrees? 

I originally asked  Joric to add this it to ble website because I said it was easier on android phone to have option to tap  -  and  +   since I also test the webpage on my old android phone and it was added :)  it's very useful on phone.  some things for soldering are just universal commands because they are useful.

[image]
it could just be a Thick font  +  and "-"  next to slide bar, it doesn't have to be that full box around it.  or  I think  a box  that is white would be better with just dark gray border on it than a full  gray background  box.

I prefer this below where box is just white with a dark grey border. it is easier to see the "+" and "-"  and cleaner. 
[image]

builder555, 2023-02-22:
i'm hesitant to implement this for 3 reasons:
1. it'll reduce the length of the sliders even further
2. there already is a way to manually enter the exact temperature
3. the upcoming "working" screen will have +/- buttons for the main controls

River-b, 2023-02-22:
dont inplement for the settings. I totally agree. keep the settings simple and smaller.
Indeed  this is only for Work screen that will also show Live temperature reading.

someone else today was tellin me they could live without the mini graph, but the Live temperature reading is a must have on work screen. many people would be happy to just have that at least :)

builder555, 2023-02-28:
closed with #48 implementation

closed 2023-02-28 by builder555