builder555/PineSAM #22: Lock down mode

River-b, 2023-02-20:
This is lower priority but, someone in ironOS was looking for a way to Lockdown all settings so students couldn't mess with the iron.  https://github.com/Ralim/IronOS/issues/1477

This is a feature of the JBC ble iron settings.
they have this for example.
[image]

and this as part of the BLE  bluetooth app for their bLe  iron which can run on the JBC display or any android/apple phone.
[image]

This might be doable on the BLE settings app. but then we need something on the ironOS side as well. and the Poster said they would be willing to collaborate with someone on the code. 
Basically it would be enabling  Lockdown mode for Pinecil  then that means you can only change the setting if you have the 3 digit or 4 digit pin number. this is good for student spaces.
Since we can enter  Type in temperature in  PineSAM already.

builder555, 2023-02-20:
To test PineSAM's usability in student spaces I'd really need to have two Pinecil's. Right now the app connects to whichever device it happens to find, there's no "selective pairing". Plus, how many student spaces actually have Pinecil V2's? I know it's an amazing device, but I don't think the market penetration is anywhere near JBC's levels.

I can implement the locking mechanism on PineSAM, but it would need to be implemented in IronOS too, otherwise it'd be trivial to bypass.

River-b, 2023-02-20:
are you in the USA?  tl_lim told me he will send a free tester ones to people doing development.

builder555, 2023-02-20:
DM'd you on discord

River-b, 2023-02-20:
I was thinking of 2 different kinds of lockdown.
One is  like a padlock symbol that you open and close with a simple click and that is just for a single user who doesn't want to  mess with they settings.  That would be more applicable to Joric's BLE for example  since if you lock-down and can still monitor  Graphs of Temp/Power while using.

Then there is the idea to set a  3 or 4 digit pin number so that one teacher could lock down settings for student maker space if some new  student just pressed all wrong buttons. or maybe no pin if there is some other way to lock it down.
They could just use the "reset to defaults"  but if they are using no-lead solder, the default 320 C  is not hot enough. 
On the pinecil itself we only have 2 buttons and Gyro- accelerometer as possible controls. therefore it seems like the BLE browser app  would be better place to type in a Pin number to adjust settings and lock-down multiple  V2 irons.  if they plugged in one at a time started PineSAM,  set up their desired settings, and  had  lock down method. 

It would not and should probably not be full-proof since maybe one day the pin doesn't work or the pin number is forgotten. One should still have the method to simply "reset to defaults" to clear it out  or simply  Flash new  Firmware again to get fresh start and not brick pinecil bc it's locked with a Pin number.

builder555, 2023-02-20:
That would be Apple's strategy - if you forget the pin, you have to buy a new Pinecil 😆 

I do see the appeal behind the lock, but as I mentioned, it would be almost pointless without IronOS integration. Or were you thinking in terms of web-only lock? i.e. students can still modify the settings on the device itself?

River-b, 2023-02-20:
There is a ticket open on IRonOS and Ralim said he was open to it but  that  it should be napkin penciled out first what could be done in Pinecil for example before doing any coding. I linked the ticket so you could see what Ralim said.

The OP  did not return so I don't know what happened.   It is very interesting that thie Pin/Lock padlock is on the JBC brand BLE app they give with their new BLE irons ($600 iron).  For big university labs that might be marketing sell point to use the JBC brand.

For maker spaces and classes, it might help.  For single home user, not so much.

builder555, 2023-02-20:
if we implement all the changes in open tickets, then pine64 can market their $60 iron (w/ usb brick) as $600 competitor 😄

builder555, 2023-04-07:
closing this issue as not planned for now. can re-visit if IronOS gets the update.

closed 2023-04-07 by builder555