builder555/PineSAM #32: Add Temperature Presets, 3 options for common fast temp changes

River-b, 2023-02-20:
would be nice to have 3 common presets for  `Set °C`
quickly tap on for the change.
I often go from Lead solder to a no-lead solder and would love to be lazy and not even scroll and pick from 3 different ones which would probably take care of most use cases.

3 boxes is good.  of course still keep the custom change feature but being able to save 3 numbers locally is nice. I suppose we would need a text file saved into the same folder as PineSAM to keep user defined settings that are accessible by the script.

[image]

builder555, 2023-02-20:
i was actually thinking to expand that into a complete preset system - you adjust all the settings, and save it as whatever you want, e.g. set temperature to 310, set sleep to 0, display to right-handed operation, call it "lead, no sleep, R-hand", create another preset for using 4S battery pack on-the-go with 3.5v limit, set sleep to 1 minute, sleep temperature to 100C call it "4S power on the go", then you just pick a preset, it sends all the settings to the device at once.

River-b, 2023-02-20:
I've requested for IronOS  to expose  the Debug information.  if this happens then  it could be shown above the temperature the tip-type.
`Short  Tip` or `Normal  Tip`  since that is what Pine store calls them. 
or  one could instead  show
`6.2  Ω profile` or `8.0  Ω profile`
these do not represent the actual ohm resistance of the tip but rather which Tip profile was selected to use.
V2 auto-detects resistance when pinecil first boots up. Ralim says it's very noisy so the Debug Tip selector  works like a toggle.
if the tip is closer to 6 ohms,  it uses the 6.2ohm profile. and if it's closer to 8.0ohms it uses the 8 ohm profile for power/temperature. 

I have 3 tips that are 7.8, 8.0, 8.3 ohm, and all three would trigger the `8.0  Ω profile`
my 6.2-6.5ohm tips trigger the `6.2  Ω profile` 

Pinecil does not have a profile for every gradation of resistance, and toggles between two profiles.

River-b, 2023-02-20:
Notice that the Slider bar can be Vertical instead of Horizontal which helps on Cell phones I think for sliding the finger and fitting stuff.

River-b, 2023-02-21:
This would be something good for the  main "work" screen   versus  settings screen.

River-b, 2023-02-22:
work screen I think should be allowed to view horizontal on phone. 
  250 -315 -375C   could be good starter presets.
if you only have room for 2 since want the buttons to be big to tap on Phone (horizontal display) and be easy to see and hit.  depends on if there is room for 2 or 3  depending on screen size.  that screenshot is from a different station where they have 3 presets on work screen for quick access.

River-Mochi, 2023-02-28:
- [ ]  2 and 3 preset buttons were made by @NeilHanlon  and have been fully tested now.

- [ ] 2 Preset buttons seems good for now and is deployed in 1.30 

- [ ] Presets Save custom temperatures: presets can be easilly altered and saved: 
1. set the desired preset using the +/- buttons, then Long hold over the Preset button you want it saved to
2. this then alters the settings.html file code. 

@builder555  I'm going to close this ticket for Presets as operational per 1.30?
 and perhaps you could open a new ticket for the Advanced Profile System you were exploring up top since that would take a bit more work for a future release?

 with Advance Profile System people could potentially have:
Profile1 = lead soldering with specific presets and other settings
Profile 2 = no lead soldering with specific presets and other settings.
Profile 3.... ext.

PR #55 (2023-02-27): Make buttons work on Work screen and various other UI things
Work screen improvements and functionality implmementation

Closes #32

- Make plus/minus buttons control set temperature as per Short/Long
  press parameters. Tapping the buttons once change the set point by
  TempChangeShortStep. Holding them changes the set point by
  TempChangeLongStep
- Work Screen changed to add preset buttons which immediately set the
  iron to the specified temperature. Holding down this button will
  prompt to confirm setting a new preset value using the current
  SetPoint.
- Presets are stored in localstora [... cut]
closed 2023-02-28 by River-Mochi