builder555/PineSAM #33: Hall Effect sensor

River-b, 2023-02-21:
As I find things while testing, I'm going to add a ticket. you can just ignore all these until  you have time/motivation :)
just want to document what I see since I'm spending time to QC it.

1. hall effect sensor (HES) should be moved to the Sleep menu section.
2. The way HES works is if you put a neodymium magnet on your stand, the moment the pinecil hits the stand, the pinecil goes to sleep mode (uses your sleep mode set temperature). 
3. This feature is on luxury pro irons and is an optional $2 add-on that all pinecil owners can solder on the U14 location on the PCB.
4. It will _only_ show on the pinecil menu, if pinecil detects that the HES was installed. 
5. Range is 0-9, the default is 7; this is a user changeable number based on how strong or weak a magnet one is using. people with a large magnet might set pinecil to 3 to prevent pinecil from going to sleep while working bc their hand was too close to the stand. I have 2 tiny neodymium magnets, so 7 works for me.
6. I suggest having it there but greyed out and unusable for people who don't have hall sensor installed or make it _only_ show in sleep menu IF pinecil detected it, like it did for one of my V2 since i haven't had a chance to add to the other V2 (see image, it shows in unused section).

order on the Pinecil screen:
- Motion Sensor
- Sleep Temp
- Sleep timeout
- Shutdown timeout
- Hall Sensor Sensitivity

[image]

They are fairly easy to install. cost $1-$2 on lcsc.com, digikey, mouser.

7. Instructions I wrote on [hall sensor in IronOS](https://ralim.github.io/IronOS/HallSensor/).
8. available [here, LCSC](https://www.lcsc.com/product-detail/Position-Sensor_SILICON-LABS-SI7210-B-00-IVR_C2654956.html). or [here, Digikey](https://www.digikey.com/en/products/detail/silicon-labs/SI7210-B-00-IV/7323047)

builder555, 2023-02-21:
is that the only compatible HES? i can't find it on aliexpress, otherwise i might have to wait until i need more stuff from digikey/mouser, their shipping costs bite.

River-b, 2023-02-21:
it's only on real digital stores like LCSC.com, digikey and mouser type stores.

River-Mochi, 2023-03-31:
- if  user has a number  in Hall sensor,  then they have a hall sensor installed.  
- Pinsam should automatically move Hall sensor from Unused Category into the  "Sleep Mode" Category as the last entry.

- On actual Pinecils, it detects if there is a hall effect sensor installed, and only if  yes, then  the adjustable Hall sensor sensitivity option appears under  Sleep Mode section.  

- Expected results: PineSAM  should do the same,  if  halleffectSensitivity number shows up, needs to appear as last item in Sleep MODE  and allow  0-9  settings  (7 is default).

[image]

commit 7e3c49da (2023-10-18): can adjust Hall Sensor setting. closes #33
commit ddbeac25 (2023-10-18): fix: can adjust Hall Sensor setting. closes #33
commit 801f21a7 (2023-10-18): Merge pull request #192 from builder555/dev

fix: can adjust Hall Sensor setting. closes #33