builder555/PineSAM #34: bug - iphone can change Set Temp with slider but not typed box.

River-b, 2023-02-21:
Expected behavior: 
changing Set temperature on slider or typed box changes temp.
Iphone behaves differently than windows PC browser.

Tested:
windows 11 Chrome browser:  works.
iphone connected to PC ip address:   Slider works and temp is changed & Powershell on PC shows Setting sent to SetTemp.  however, oddly typing in the blue box does not work.  nothing happends on Pinecil or WP (windows powershell) terminal.   it also lets me type "700" in the box but that doesn't matter since it's not changing anything.

builder555, 2023-02-21:
while i'm looking for that android phone to test on, can you check if your iphone now works correctly?

River-b, 2023-02-21:
- Iphone works now with both slider and pop-up text box.
- I had to clear data to get new display on iphone to show.
- button size is good on iphone like the Show hints and  save to flash green button, button could stay same or even be a little smaller.

closed 2023-02-21 by River-b