builder555/PineSAM #39: Display  Input Voltage

River-b, 2023-02-21:
1. Display

- Input voltage (example 20.2V or 28.0V)
- handle temperature (Han-C) if your handle (temperature on your PCB inside handle) looks like it's 50C, probably good idea to turn off pinecil and let it cool off for example.  Good Safety information.

these are already exposed in BLE. Would be nice if they were part of a main screen with the Set °C  ( I think it's easier instead of saying Set Temperature to just say  Set °C  or Set °F  depending on what people selected for C or F, and they will know  what that means without the longer word  Soldering Temp). 

2. add a category called `Debug data`  to go before "Unused" which will show Tip R, Tip Res, and other information which is  not changeable by the user but very helpful in diagnosis and troubleshooting.  when people come into discord with a problem, we often ask them to give us the R-Tip and C-han right away since that is indicator of a possible Temperature sensor or Mosfet problem.  I believe that some of the Debug info is already transmitted.

builder555, 2023-02-28:
Since voltage is already visible on the work view, this ticket could probably be a proper "debug view".

ACs:

- [ ] display all available "live" values (see #48)
- [ ] refresh values once every 5 seconds (to prevent bt/cpu stress)

River-Mochi, 2023-02-28:
I am separating out  Debug information as most of  that is not some thing people need to see in the Work view.
Except for handle temperature  (which could also wait to go with graph view if there is no room for handle  temperature in the Work view screen?  depends on what you think for fitting it in. 

I'm opening a separate ticket for Debug info.

commit 98488fab (2023-03-14): feat: add debug data. closes #39
closed 2023-03-14 by builder555