builder555/PineSAM #40: scaling issue on iphone

River-b, 2023-02-21:
**To Reproduce**
Steps to reproduce the behaviour:
1. open in iphone, chrome browser, 1 letter will be cut off and not visible on the screen.

**Expected behaviour**
that there is a little white space on the left side of the screen before seeing words and that no letters are cut off. 

**Actual behaviour**
1-2 letters are cut off on the left side, other big fonts like PineSAM are kissing the very edge.

**Screenshots**
[image]

**Specs:**
 - OS: iphone
 - Browser: chrome
 - **PineSAM** Version: 1.0.13 
 - IronOS build: beta 2.20.EB5D6EA  18-02-23

River-b, 2023-02-22:
windows 11 chrome example 
Needs more white space border on the PC browser too.  if I make windows browser smaller to just 2 categories or 1 category view, then it also cuts off letters on the left side, and you can see right side is tight too, Show hints gets cut off and blue boxes are close to edge. 
[image]

builder555, 2023-03-13:
[image]
seems to have been fixed with the latest layout changes

closed 2023-03-13 by builder555