builder555/PineSAM #48: add a "Working" screen

builder555, 2023-02-22:
The screen should:

- [ ] display current temperature
- [ ] display current wattage
- [ ] display max wattage
- [ ] display current voltage
- [ ] display max voltage
- [ ] be able to change set temperature

River-b, 2023-02-22:
ty for making this ticket much cleaner and more organized

builder555, 2023-02-23:
can read:

LiveTemp
SetTemp
Voltage
HandleTemp
PWMLevel
PowerSource
TipResistance
Uptime
MovementTime
MaxTipTempAbility
uVoltsTip
HallSensor
OperatingMode
Watts

but reading all of them slows down the device and BLE communication. for the sake of speed and better UX, only these values will be returned:

LiveTemp
Voltage
HandleTemp
OperatingMode
Watts

River-b, 2023-02-23:
Sounds good. most of those are DEbug info that would not need to be displayed in the Work view,
I was thinking maybe add an extra section to Settings called "Debug Data"
and put most of that in there. some of it actually doesn't change and is static . 

if you don't put it on Work View and it's on another Settings that does't ask for it until we get to that section does that help? 
We  need: Set temp, LiveTemp, handle temp, watts for sure.

I dont know what VoltsTip is. 
TipResistance is a one time static number it picks either 6.2 ohm or 8 ohm profile when Pinecil first boots depending on the tip you have installed at initial boot since it only reads at boot. once you get that you dont have to get it again unless pinecil is rebooted. so that tells us what kind of Tip Profile is being used but doesn't need to be constantly requested.

River-b, 2023-02-23:
"uVoltsTip"
[T] <Ralimtek> micro-volts raw reading of the tip.
Same as in debug menu

I think this is R-Tip then per documents [debug details ](https://ralim.github.io/IronOS/DebugMenu/)
and would only be needed for Diagnostics and troubleshooting and not for active view.

River-b, 2023-02-23:
will hall sensor still be in the Sleep settings sector?

builder555, 2023-02-23:
> will hall sensor still be in the Sleep settings sector?

that's for #33

River-Mochi, 2023-02-26:
For reference
1. add  Double  Arrow icon to "Peak Watts" that Builder found. 

- either:   [skinnier one](https://www.iconfinder.com/search?q=double&iconset=font-awesome)
- or: [Thicker](https://www.iconfinder.com/icons/8541581/angle_double_up_icon)
Skinny one might look better, needs testing.

builder555, 2023-02-28:
as per our discussion, removing "max voltage" from acceptance criteria

closed 2023-02-28 by builder555