builder555/PineSAM #53: Toogle button for Calibrate CJC has a different shape

River-b, 2023-02-25:
**To Reproduce**
open Chrome browser on windows PC

**Expected behaviour**
all toggle buttons are the same shape and size

**Actual behaviour**
Only the Calibrate toggle button is strange shape. it looks like instead of a white ball, it is vertical and partially cut off

**Screenshots**
[image]

builder555, 2023-02-25:
ah, windows edge cases. i'll tweak the switches style a little

River-Mochi, 2023-03-10:
fixed now  in version Source Code  PineSAM-1.5.7 I just tested.

[image]

closed 2023-03-10 by River-Mochi