builder555/PineSAM #54: graph display option

River-b, 2023-02-25:
**Is your feature request related to a problem? Please describe.**
Future posibility and  could enlist others to help with this.

have a graph feature that can be toggled on or just on all the time.

- Left Y=Temp, Right Y=watts, bottom X = time
- 2 lines tracking temp and watts, (temp = orange, watts = different color blue or purple).

Why?
because people love graphs. it may also help to show how well Pinecil is performing if there are not huge slumps in temperature during soldering.
(a bit esoteric, but people could record their PC screen and play it back later to see if they need to modify their tip or power for better performance and less temperature drop during soldering of particular parts; similar to charts in pro expensive irons).

River-Mochi, 2023-03-05:
sample Graph view
[image]

River-Mochi, 2023-03-31:
per 1.12.11 version.

-  thinner graph lines for Temperature and  Watts. right  now they are thick and not as smooth looking as other graphs.
- add veritical  lines for 1 sec/div  interval
- all graph lines should be "dashed"  and very  light gray  to not overwhelm the Temp/watts line colors
-  switch to FF00BF  for the  Temperature line as it compliments PineSAM colors and works
- X bar,  not in  bottom right   "Time 1s/div"

builder555, 2023-04-02:
broke out some points into their own tickets, otherwise this issue is an entire project on its own.

commit 0a3f07fb (2023-04-02): fix: added vertical grid. closes #54
closed 2023-04-02 by builder555