builder555/PineSAM #66: change icon to reflect current temperature

builder555, 2023-02-28:
- [ ] low-temperature icon for temperature <50C
- [ ] mid-temperature icon for 50C < temperature <=100C
- [ ] high-temperature icon when current temperature >100C

[low temp icon](https://fontawesome.com/v5/icons/thermometer-quarter?s=solid&f=classic)
[mid temp icon](https://fontawesome.com/v5/icons/thermometer-half?s=solid&f=classic)
[high temp icon](https://fontawesome.com/v5/icons/thermometer-three-quarters?s=solid&f=classic)

River-Mochi, 2023-03-04:
Love the idea of  3 different Temp Icons that change depending on the range of temperatures.

Is it also possible for these icons to get a color? Physical shape change will help people with color  issues.
for people who can see color perhaps:

low temp icon  = make icon itself  yellow  (<50C)
mid temp icon  = orange   (50C <=200C)
high temp icon = Red  (> 201C)

- using alert colors  like typical icons use   caution = yellow, and stop or Red alert/danger  = red

River-Mochi, 2023-03-04:
Could there be 4 icons to represent live temperature?

commit 72d0c870 (2023-03-13): feat: colour and icon change with temperature. closes #66 and #57
closed 2023-03-13 by builder555