builder555/PineSAM #67: Release page enhancements

River-Mochi, 2023-02-28:
**Is your feature request related to a problem? Please describe.**
**Releases page**

- [ ] 1. Is it possible to change the name of the downloads from generic "Source code"  name to 

something like PineSAM.zip or PineSAM-1.3.1.zip. It currently says "Source Code" which is a little confusing to some people who want to download PineSAM. There is a tag for version, but even having it with repo PineSAM name would help.

reference: 
-https://docs.github.com/en/repositories/releasing-projects-on-github/managing-releases-in-a-repository
- https://github.com/marketplace/actions/auto-release
- https://github.com/marketplace/actions/automatic-releases

[image]

Example
[image]

- [ ] 2. add a **Contributors** section to Releases page, similar to how the Icons show for [IronOS releases](https://github.com/Ralim/IronOS/releases)
- [ ] 3. add a New contributors section, this is for people who are making their first contribution to the repo, [ example here ](https://github.com/Ralim/IronOS/releases)

builder555, 2023-02-28:
1. that is related to a new build process, i.e. packaging the application for release (e.g. using pyinstaller), i added #69 to address that

2, 3. Need to modify the release workflow for it, I need to research best/easiest way to implement it.

River-Mochi, 2023-02-28:
> research best/easiest way to implement it

I did a view page code on Ralim's "release" page to see how he is adding contributors to display their Avator icons, but whole page is a little complex so I thought it was just better to link to you IronOS release page and you can view page code directly.

builder555, 2023-02-28:
it's probably automatically generated by one of the workfows in .github/workflows

River-Mochi, 2023-03-04:
This ticket  now  will also  fix the problem with Downloads badge not working.
once there is an archive there that is a binary zip package with a unique name  (not Source Code), then the API page for releases should start showing  download_count and that is what the badge reads to get count.

 it's probably ok to leave the 2 source code zips there, as many other repos just leave it there.  and have the new name package which will show up at the top.  most people  probably know from install instructions or  they are used to going in github and grabbing the thing with 'real name'  versus scary thing called 'source code.zip'

builder555, 2023-03-13:
done

closed 2023-03-13 by builder555