builder555/PineSAM #72: change follder name to PineSAM

River-Mochi, 2023-03-01:
**Is your feature request related to a problem? Please describe.**

- Consider just naming the internal folder **PineSAM**
- Currently it looks like this:  `PineSAM-master1.3.1\PineSAM-Master\backend`
- I like that the main folder name has the version number; for ease of instructions, the internal folder of all downloads could simply be PineSAM.
- new structure would look like:  `PineSAM-master1.3.1\PineSAM\backend`  or  `PineSAM-master1.3.1\Pinesam\backend`

[image]

builder555, 2023-03-01:
I'm not sure what you mean, currently if you download from the release, it extracts into `PineSAM-1.3.1`
[image]

River-Mochi, 2023-03-01:
I mean the second level folder name and not the highest level.  see the screenshot image I included.
instead of **Pinesam-master**\backend  change it to just be pinesam\backend and remove the word "-master"
for what the user sees when they unxzip it.

technically they will see 3 folders/directories
PineSAM1.3.1\Pinesam\backend

builder555, 2023-03-01:
that's what i'm referring to as well, when i unzip it, i only see `PineSAM-1.3.1\backend` i don't have the middle folder "pinesam-master" or anything else

River-Mochi, 2023-03-01:
I see the Problem  the kink from the Windows Readme instructions gives a completely different file directory than the  Releases page.

[image]

builder555, 2023-03-01:
done

closed 2023-03-01 by River-Mochi