builder555/PineSAM #80: Graph:  record and play log of temperature/watts

River-Mochi, 2023-03-05:
**Is your feature request related to a problem? Please describe.**

request was made for recording feature of the graph log to be able to play it back for analysis of soldering.

**Describe the solution you'd like**

- [ ] record and save log
- [ ] way to store logs 
- [ ] Graph full view has a media Play/Stop button to play back the log of Temperature/Watts graph.
- [ ] Forward and Reverse button.

**Additional context**
For the short term, suggest using one of the screen recorders to record the screen with the graph activity for playback and inspection.

River-Mochi, 2023-03-05:
### Sample graph view with media play buttons
[image]

builder555, 2024-01-28:
What would be the practical use case for it?

builder555, 2024-01-31:
Given the lack of responses for this feature, I'm closing it as not planned. If there is a renewed interest, I'll re-open.

closed 2024-01-31 by builder555