builder555/PineSAM #84: Notify if a new IronOS Release firmware is available

River-Mochi, 2023-03-08:
**Is your feature request related to a problem? Please describe.**
**Background**

- It is a good idea to keep Pinecil updated.
- Pinecil owner's don't always know when there is a new release. some releases have important bug fixes and work arounds for compatibility with more chargers.
- Certain menu options that are taken for granted in new releases don't exist on older versions so there can be communication problems when troubleshooting.
- Some firmware updates resolve issues and prevent new ironOS tickets.
- This was never possible before when we did not have a BLE connection in pinecil.

**Describe the solution you'd like**

- [ ] send current installed firmware number over BLE to PineSAM (IronOS  support needed).
- [ ] Step 1, pinecil displays the installed firmware version in the collapsable references section (i.e. potentially same place as ID number, serial number) for things we don't need to see every day to work.
- [ ] Step 2. PineSAM checks the api Release page of IronOS to compare if the release number is higher than the current, and there is an icon that appears that lets the user know that an new stable update is available.
- [ ]  "update available" icon could be at the top of the screen, maybe glowing, a rainbow or happy face :)  jk.

River-Mochi, 2023-03-24:
FYI  this just went up

Add FW version to BLE characteristic #2801
https://github.com/Ralim/IronOS/actions/runs/4497563158

builder555, 2023-04-09:
closed in 095f7c0181e67ae865caa0dbf20b1b74d61420d5

closed 2023-04-09 by builder555