builder555/PineSAM #85: Notify users if a new stable PineSAM release is available

River-Mochi, 2023-03-08:
**Is your feature request related to a problem? Please describe.**
Notify users if a new stable PineSAM release is available so they don't have to keep checking in on the github weekly.

**Describe the solution you'd like**

- [ ] Notify users if a new stable PineSAM release is available
- [ ] perhaps just check one time when the app is initially opened and compare it to the version already running.
- [ ] have a unique small icon at the top of the Work View that appears and stays there as long as they are on older version.  if they click on the icon, it displays "New PineSAM version available"
- [ ] also when users first start the script, the Browser has a Pop-up that alerts one time at the startm, in a friendly color, "New PineSAM update available, click on the Github Cat in the corner".

River-Mochi, 2023-03-14:
For this,  was thinking only notify for major release change and not the little ones?  since the little ones seem to come several times per day.  it could get a little spammy.

builder555, 2023-03-14:
i'm making it like this. very non-intrusive 
[image]

River-Mochi, 2023-03-14:
and then we just Click on the CAT  to go and get it?

builder555, 2023-03-14:
you can click on the cat or on the "New version available" it's a link also

closed 2023-03-14 by builder555