builder555/PineSAM #101: warnings when running  setup-dev.bat on windows 11

River-Mochi, 2023-03-13:
windows 11, powershell.
this warning appears every time I use setup-dev.bat

[image]

1.  since I already have python installed  is there a reason it can not find it? or is this an error message that can be ignored?
2. "WARN deprecated sourcemap-codec@1.4.8: Please use @jridgewell/sourcemap-codec instead"  
does this need to be changed in the script to something else so the warning can go away?
[image]

River-Mochi, 2023-03-22:
I have part of this figured out.

On windows, the python executable is  simply _python_ and not python3  ([Reference/Read this])(https://superuser.com/questions/1576758/how-do-i-alias-python3-on-windows)

1.   ` .\setup-dev.bat `   will get python not found  because  typing "python3"  doesn't work in the Terminal since there is no python3.exe    or alias.
I changed  setup-dev.bat  to  just say "python"  with no "3"  in two spots and now it runs completely differently

2. Either  this BAT  script needs to change for Windows users or instructions need to be included on how to may alias for python3 --> python

3. If  windows user is using Powershell then they can make a "temporary" alias  with 
`Set-Alias -Name python3 -Value python`
to make it permanent is a lot more steps - see that reference link above  there are 3 different methods, all kind of complex  for users so maybe change this file to  call for "python" and not "python3" 

```
@echo off
where python >nul 2>&1 || (
    echo python could not be found
    exit /b
)
where npm >nul 2>&1 || (
    echo npm could not be found
    cd ..
    exit /b
)
cd backend
python -m pip install --user bleak websockets requests
```

### With python3 changed to python, this is the new output with no python errors

[image]

River-Mochi, 2023-03-22:
I still get the warning about  "Deprecated Sourcemap-codec"  advising to use some other package  but I'm not sure  what to do about that.   is that something you have to do on PineSAM side or something  I have to install on Windows?

builder555, 2023-11-22:
These warnings can come up when installing npm packages. In this case it's not an issue, since that package version(1.4.8) was tested with this version of PineSAM and it works. It would be very time-consuming to update the application with every new npm package release, so unless it causes a problem it's probably best to just ignore them.

closed 2023-11-22 by builder555