builder555/PineSAM #110: IronOS  changed  to 128 UUID

River-Mochi, 2023-03-14:
Change to 128bit UUID's #1608  is now  merged into IRonOS  beta ( coming in the next release)

Please check that it doesn't break PineSAM

https://github.com/Ralim/IronOS/pull/1608/files

builder555, 2023-03-14:
@River-Mochi could you paste a link to the FW with these changes? it'll be easier for me to test against.

River-Mochi, 2023-03-14:
https://github.com/Ralim/IronOS/actions/runs/4420347769

at bottom are the Zip filles for Pinecil V2

builder555, 2023-03-15:
this version does not display any services. needs further investigation on the firmware side.
Used this test code:

```
from bleak import BleakClient, BleakScanner
import asyncio
async def get_all_ble_services():
    devices = await BleakScanner.discover()
    for d in devices:
        if not 'pinecil' in str(d.name).lower():
            continue
        async with BleakClient(d.address) as client:
            await client.connect()
            print('connected to ', d.address, d.name)
            for service in client.services:
                print('Service:', service.uuid)

asyncio.run(get_all_ble_services())
```

River-Mochi, 2023-03-26:
@builder555 

Ralim announced  128 UUID is now ready for testing - some APP  changes will be required for it to work.
Download beta bottom of this page:
**https://github.com/Ralim/IronOS/actions/runs/4522384503**

I "think"  this one might also include the  Firmware IRonOS version number since this only cable today after the FW  fix.

builder555, 2023-03-26:
Tested - the new FW does not return any values, only 0xffff

Spagett1, 2023-03-30:
Ive tested the latest release of pinesam with the following firmware with no issues. 

https://github.com/Ralim/IronOS/actions/runs/4549363895

Im running artix linux and my browser is firefox.

River-Mochi, 2023-03-31:
I've tested  it also  and  have Bug on Screen  brightness change.

@ spagett1  please check  Screen brightness on  on new Pinecil,  should only be 1-5 option.  and also check in linux PineSAM.

 on PineSAM   do you see weird numbers?    also pineSAM should limit you to only 1-5 range if you are on this newest 128-uuid  ironos.

builder555, 2023-04-07:
fixed in #f28bc0db97ee072d04d406d6fc8a252a4803d949

closed 2023-04-07 by builder555