builder555/PineSAM #121: Show  BLE MAC address

River-Mochi, 2023-03-22:
**Is your feature request related to a problem? Please describe.**

In the teminal (e.g., powershell),  the script nicely shows Bluetooth BLE MAC address  of the pinecil it finds; this is different from  BLE Unique name.

This info would be useful for people using ESP Home assistant and other possible uses.

**Describe the solution you'd like**
Is it possible to display  BLE MAC  address somewhere in PineSAM - doesn't need to be too obvious.

possible locations: 
1. Debug category somewhere as it's not info normally needed
2. under PineSAM  logo  so that  both 'BLE Unigue name' & MAC show (MAC could be in smaller light gray font next to Unique name).
MAC address actually may be more useful than Unique name in the future.

**Alternatives**
In wiki  usage instructions,  I could add how to find BLE MAC address in the terminal display which will be at "Found pinecil"
it's not that hard to just go to the terminal, but if you are on your phone, it's  convenient to have it listed on the HTML somewhere.
[image]

builder555, 2023-03-26:
currently on MacOS it's very difficult to get the MAC address with the current setup. if this info is sent in one of the BLE services from IronOS, that'll make it much easier to do.

builder555, 2024-01-28:
Closing this issue until IronOS can send mac address. At the moment only ID is sent.