builder555/PineSAM #125: Stop using external resources

MagicLike, 2023-03-26:
**Is your feature request related to a problem? Please describe.**
Especially Google and Cloudflare are absolutely privacy invasive.
Cloudflare can among other things:

- inject / modify JS (into the website)
- spy on you and the page's users
- track users across websites 

Google (and Google Fonts in this case) tracks users across pages and spies on what they are doing too. Also Google Fonts is - to my knowledge - illegal to use in the EU, due to the GDPR. 

**Describe the solution you'd like**
Please host such external resources - like CSS and fonts - locally, rather than calling them every time.
Another option for Google Fonts would be to use [Bunny Fonts](https://fonts.bunny.net).

**Additional context**
[image]

builder555, 2023-03-26:
@MagicLike is using google fonts illegal only when they're hosted on google's CDN or even locally-hosted?

MagicLike, 2023-03-26:
> @MagicLike is using google fonts illegal only when they're hosted on google's CDN or even locally-hosted?

Nope, as if you are "hosting Google Fonts locally" you are only hosting the font files and the license (typically something like OFL.txt), so you can't really call it hosting Google Fonts... ^^

commit 72536cf8 (2023-03-26): fix: store external dependencies locally. closes #125
closed 2023-03-26 by builder555