builder555/PineSAM #131: White Screen - No PineSam

LtanHonor, 2023-03-29:
While attempting to setup PineSAM, everything appeared to go well.  

*Current Versions*

Python 3.10.9 <---> version 3.11 has been used as well
pipenv-2023.3.20
Node-v18.15.0-x64
Browsers Used : Chrome | Firefox | Vivaldi | Brave

*setup-dev.bat results:*
PineSAM [   ﱔ     ]
ᴥ .\setup-dev.bat
Requirement already satisfied: bleak in c:\users\ltanh\appdata\roaming\python\python310\site-packages (0.20.1)
Requirement already satisfied: websockets in c:\users\ltanh\appdata\roaming\python\python310\site-packages (10.4)
Requirement already satisfied: requests in c:\users\ltanh\appdata\roaming\python\python310\site-packages (2.28.2)
Requirement already satisfied: bleak-winrt<2.0.0,>=1.2.0 in c:\users\ltanh\appdata\roaming\python\python310\site-packages (from bleak) (1.2.0)
Requirement already satisfied: async-timeout<5,>=3.0.0 in c:\users\ltanh\appdata\roaming\python\python310\site-packages (from bleak) (4.0.2)
Requirement already satisfied: idna<4,>=2.5 in c:\users\ltanh\appdata\roaming\python\python310\site-packages (from requests) (3.4)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in c:\users\ltanh\appdata\roaming\python\python310\site-packages (from requests) (1.26.15)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\ltanh\appdata\roaming\python\python310\site-packages (from requests) (2022.12.7)
Requirement already satisfied: charset-normalizer<4,>=2 in c:\users\ltanh\appdata\roaming\python\python310\site-packages (from requests) (3.1.0)
npm WARN deprecated sourcemap-codec@1.4.8: Please use @jridgewell/sourcemap-codec instead

added 268 packages, and audited 269 packages in 10s

50 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

*run-dev.bat results:*
PineSAM [   ﱔ     ]
ᴥ .\run-dev.bat
20:25:10.498::INFO::Starting pinecil monitor
20:25:10.498::INFO::waiting for pinecil...
20:25:10.500::INFO::Detecting "pinecil"...
20:25:10.553::INFO::server listening on 0.0.0.0:12999

> vue-project@0.0.0 dev
> vite --port 8080 --host 0.0.0.0

  VITE v4.1.4  ready in 418 ms

  ➜  Local:   http://localhost:8080/
  ➜  Network: http://192.168.4.100:8080/
  ➜  press h to show help
ERROR: The process with PID 0 could not be terminated.
Reason: This is critical system process. Taskkill cannot end this process.
ERROR: The process "0" not found.

PineSAM [   ﱔ     ] took 2s
❯ 20:25:15.567::INFO::Found pinecil at C4:D7:FD:D7:4F:91

Then the webpage just sits at a white screen with PineSAM as the tab label.

builder555, 2023-03-29:
did you try the compiled version?

LtanHonor, 2023-03-29:
Same result.  Not sure what is messing up.  The Pinecil V2 works with Joirc's web app.

builder555, 2023-03-29:
do you see any errors in terminal? it sounds like python backend is crashing.

LtanHonor, 2023-03-29:
I will check in a bit.  This is also on Windows 10 if that makes an anthill.

builder555, 2023-03-29:
if you have some time, try this:
open 2 terminal windows, navigate to pinesam directory in both, then in one:

```
cd backend
python main_server.py
```

and in the other one:

```
cd ui
npm run dev
```
then you should be able to open http://localhost:8080
and see if either window has any errors. I don't have easy access to a windows machine, so if you see something suspicious, it'll help me debug the problem.

River-Mochi, 2023-03-29:
on Bright note,  at least on Windows10,  Ltan  doesn't have the  additional "access denied" to esbuild.
They were saved one error message 😃 

maybe something with extra security in win11.  don't know what esbuild does to fig it out yet.
`node_modules\@esbuild\win32-x64\esbuild.exe - Access is denied.`

builder555, 2023-03-29:
@LtanHonor do you mean blank page as in there's absolutely nothing on it? or blank as in no data, i.e. like this 
[image]

LtanHonor, 2023-03-29:
[image]

builder555, 2023-03-29:
ok it looks like something is missing in the OS, blank screen means that UI couldn't start at all. 
the error you're seeing on the backend side indicates that the script can't communicate with pinecil, also a weird error since it's able to find it. 

unfortunately without access to a windows machine, I can't help much further.

River-Mochi, 2023-03-30:
for me,  either  a permission was wrong or  some needed  app was not installed; I have Windows 11 though.

**LeCaNo** from pinecil chat  has Windows 10   and got  it working  (try asking them in chat to help as they will have more similar setup)  

since I have Windows 11,  and I installed a different python 3.11  I'm not sure. I've never had it be just an all white screen in the browser.

River-Mochi, 2023-03-30:
- Since you are stuck  could try completely Fresh steps. 
-  not sure all of these apps are needed to get it to work on Windows DEV version (Source-all-v___.zip  file).  But, I went ahead and installed all of them since builder555 listed them as required and installed in this order: 
-  **git, python, pipenv, nodeJ/npm**   (exact steps I took are below).

Steps I followed exactly: 
1.  install git   (did this first)   https://git-scm.com/book/en/v2/Getting-Started-Installing-Git
2. Remove **all** old versions of python completely (add/remove programs).
3. Delete all python entries in the system variables path, go to windows run box, type "edit the system environment"  click on the shortcut it gives,  enter Environment Variables, System variables.
[image]

4. Install fresh new python using all the options checked exactly as [shown here](https://github.com/builder555/PineSAM/discussions/7)  . Download [link here ](https://www.python.org/downloads/)

5. I followed these exact steps  to install  Pipenv for windows:  https://www.pythontutorial.net/python-basics/install-pipenv-windows/

6.  Remove  nodeJS.   Reinstall it [from here](https://nodejs.org/en/download). Accept **all** other tools/packages it prompts/asks to install.  I also accepted the Chocolatey package which popped up in Terminal window during Node+npm install.  This is the terminal that pops up  when you do the NodeJS install and accept to install all the tools.  Terminal says to Press key to install the tools (see image) and then it starts installing.
[image]

closed 2023-03-31 by LtanHonor