builder555/PineSAM #133: Wrong screen brightness values and  pre-made Windows binary shows "181818181818"

StevenSeifried, 2023-03-30:
**To Reproduce**
Steps to reproduce the behaviour:
1.  Download the pre-made Windows binary, extract and start it
2. Open PineSAM in Chrome or Firefox
3. Scroll down to "Screen brightness"

**Expected behaviour**
- "Screen brightness" can be changed from 1 to 5 (was changed recently in [PR 1620](https://github.com/Ralim/IronOS/pull/1622))
- Setting doesn't show "181818181818" 

**Actual behaviour**
- "Screen brightness" can be changed from 1 to 10
- Setting show "181818181818"  at first start
    - Moving slider fixed this

**Screenshots**
[image]

**Specs:**
 - OS: Windows 10
 - Browser: Chrome and Firefox
 - PineSAM Version: 1.12.9
 - IronOS build: [e.g. 2.20.E13F118]

**Additional context**
Only pre-made Windows binary affected, pre-made Ubuntu binary works as expected

River-Mochi, 2023-03-30:
1. reminder that screen brightness code changed in IronOS and is included with the newest beta  128UUID  that Steven tested.  it went from 1-10  into 1-5  (default 2).  People with  2.20 version will still have 1-10 (default 4)

2.  Steven, thanks for testing on Ubuntu.  are you saying you got no bugs using linux  pre-made and this only happened using  windows pre-made binary?

3. would you mind also testing on a phone.  fairly easy ,  start it on windows first and/or linux.  let the terminal script keep running.  find IP address of PC.  go to phone  any browser,  enter  http://IpAddressofMyPC:8080/

4. should be able to do all the same things in Phone now. Phone  needs to be within wifi  network that PC is on. and Pinecil needs to stay within bluetooth range of the PC that is running the python script.  Could close the PC  browser or leave it open but it might get confusing if changing settings in phone since it doesn't auto-reload the other browser unless you F5 it. 

5. no one has tested this yet but  seems like if one loaded python onto a phone, it could run natively on the phone. however I'm really fine just doing the easy steps of  going to 8080 port of my  PC.

StevenSeifried, 2023-03-30:
> Steven, thanks for testing pre-made binary Linux on Ubuntu. are you saying you got no bugs using Linux pre-made and this only happened using Windows pre-made binary?

Small correction: I tested the Ubuntu/Linux binary under Arch Linux. Sry, my fault that I didn't mention this.

The bug also shown on Android (tested with Firefox and Chrome), but with `363636363636` instead of `181818181818`

[image]

River-Mochi, 2023-03-31:
Confirmed I also have this  bug using  Windows binary 1.12.11

- bug1:  weird long number shows on screen brightness
- bug 2:  slider  allows me to set 1-10 even though the firmware I'm only only allows Pinecil to go for 1-5 (with 2 as default)
- bug 3:  any number I type into the purple box shows a weird long number.  it also  allows me to enter bad numbers like 15 which are out of range and then also gives a weird long number over the whole box area.

[image]

builder555, 2023-03-31:
since this is still a beta version of IronOS undergoing rapid changes, I won't update PineSAM just yet. Once it's stable, if the issue is still present, I'll look into it.

River-Mochi, 2023-03-31:
- It happens for both V2 I tested.   and for Stephen and some other people.  IronOS   Changed the way it handles  Screen Brightness a couple weeks ago and  PineSAM  simply did not update for  this change yet which is why I think we are seeing this error across the board. 

- The Screen brightness fix  in IronOS  is completely done and ready and I doubt it will change. it will be released with 2.21  very soon.  Pinesam  needs to change to be  only range of  1-5  for people on 2.21 

- all 2.21  Milestones are **closed** and   there will be no more tickets added to 2.21,  they will all go into  far future 2.22 release for several months from now.  see here for tracking the milestones. (https://github.com/Ralim/IronOS/milestones)  

-  the only change that is happening now  is related to the 128 uuid  working or not working fixes.   the Screen brightness  change is set and done in IronOS and will not have any rapid changes I don't think related to screen brightness.
- once  we confirm  that the 128 uuid  all works  I think  2.21 will be released very soon.  it will have this Screen Brightness change   for a range of  1-5   and   got rid of the 1-10 range. 

-  this  128 uuid  version is basically the "end of the road"  and stable 2.21 will be released soon.  no more things will be added to it.  unless we find something broken in the Bluetooth 128 uuid part I think.

River-Mochi, 2023-04-07:
Recommend  changing  Brightness from slider to drop-down menu where only  1-5 can be selected just like the other similar items like battery cell size where a range sliders is not needed as there are no  increments beteen the whole number values.
1
2
3
4
5

people on older 2.20   should have 1-10.

River-Mochi, 2023-04-07:
Code change in Ralims's IronOS is here with new Steps that makes this 1-5 steps. 

I tried to implement this in  my fork and  it kind works.  I didn't get whole numbers 1,2,3,4,5  all of them were decimal points.   which will look odd if this  becomes a drop down box. 

https://github.com/Ralim/IronOS/blob/9c8abbdfc21445f39bcf0534c508248fe2015248/source/Core/BSP/Pinecilv2/configuration.h#L63

```
 * OLED Brightness
 *
 */
#define MIN_BRIGHTNESS             1   // Min OLED brightness selectable
#define MAX_BRIGHTNESS             101 // Max OLED brightness selectable
#define BRIGHTNESS_STEP            25  // OLED brightness increment
#define DEFAULT_BRIGHTNESS         26  // default OLED brightness

```

builder555, 2023-04-07:
yup, i'm working on it

commit 3f44fc5e (2023-04-07): fix: display proper brightness values for v2.21. closes #133
closed 2023-04-07 by builder555