builder555/PineSAM #147: main_server crashes if older version of IronOS  is on pinecil

River-Mochi, 2023-03-31:
**Is your feature request related to a problem? Please describe.**

1. I have 2 different V2 on my desk, one with  new IronOS128-uuid ble beta and one with old ble beta ironos. 
2. if you  grab the wrong pinecil that doesn't have newest 128UUID  ironos,   then` main_server`  terminal crashes immediately (serve.exe terminal stays open)
3.  Is Pinesam not backwards compatible then with older beta BLE versions  (many out there), which is fine too since 2.21 is the  first official  release with real BLE on it.  probably best to have people update to newest BLE since the old BLE labels are not standardized.

**Describe the solution you'd like**

4.  any way to add an Error message to the Browser screen that opens  and/or  in the terminal  where the the c:> .\start.ps1 script  is started
`This version of IronOS is not compatible with PineSAM, please update pinecil firmware`

River-Mochi, 2023-04-02:
Since I deleted the old   beta that was crashing  main_server, I tried all of these   beta's last night and none of these cause  Pinesam main_server to crash so I can not reproduce it  using pinesam  1.12.11

these beta versions all tested last night and  none of these cause  pinesame main_server to crash. I ran out of time so I can't find the one that it was.

1.  works 26-03-23  
- https://github.com/Ralim/IronOS/actions/runs/4521967996

2.  works 25-03-23
- https://github.com/Ralim/IronOS/actions/runs/4521748756

3.  works 20-03-23
- https://github.com/Ralim/IronOS/actions/runs/4472863650

4.  works 18-02-23
- from joric wiki page https://github.com/joric/pinecil/wiki

builder555, 2023-04-09:
without a firmware to reproduce the issue, i can't fix this.

closed 2023-04-09 by builder555