builder555/PineSAM #162: can't get temperature to stop going down to lowest setting

River-Mochi, 2023-04-05:
**To Reproduce**
Steps to reproduce the behaviour:
pinesam  app buttons +/-  continue to go  up  or all way down. 

android phone screen goes all way up to Max temp still but now it's also doing it in reverse Minus and going all way to lowest.   before it was only doing it to maximum (+)  now it's also doing it to minumum then get warning  out of range (10-450). 

Tested two ways.  This happens even if I  start browser fresh, and do not touch any Preset buttons first.  I just start changing the temperature with +/- screen,   Before it happened on both iphone and android and now the iphone seems ok.  I tested by  not even touching any Preset first and just using the  +/-  buttons; it shoots all way  up if I preset (+).  then I restart browser fresh  without restarting the  servers, and  press only  (-)  and it shoots all way down without stopping, even if I then try to hit the preset then it will temporaraly go to preset number and start  moving down rapidly again (following my increments of 10 from settings). 

New behavior is now it's happening on minus  as well. 

**Specs:**
 - OS: [ Windows 11]
 - Browser: [chrome]
 - PineSAM Version:  1.12.14    
 - IronOS build/date: [ 02-04-23 , 2.20.9C6ED68]

**Additional context**
related to this issue https://github.com/builder555/PineSAM/issues/119

River-Mochi, 2023-04-05:
new test:
Windows  browser it is all good. 
 iphone browser and it's all good.
   
- This is only happening on android phone, and happening for both +/-   .  restarting android chrome browser does not fix it. 
- happening on both Chrome and Samsung Internet browser on android. 
- restarting android does not  fix.

River-Mochi, 2023-04-05:
If  we can't get this fixed I'm willing to mail you this Android phone  if it's not too much shipping. this is  my spare phone that  is not in sevice that I use just  displaying  PineSAM or Joric API  while soldering.  it sits in a stand on the side so it's not getting harmed at all.

builder555, 2023-11-22:
@River-Mochi can you check if this is still the case? I believe it was addressed in one of the earlier releases.

closed 2023-11-30 by builder555