builder555/PineSAM #163: change toggle for save changes to flash

River-Mochi, 2023-04-05:
Change  app toggle to option 1 or 2.

| original | option 1 |option 2|
| :----------------- | :-------------- | :--------------|
| Save changes to flash | Save to flash  | Save to pinecil|

It is clear, shorter, cleaner on the screen.  
People have "show hints" if they want the full words.

River-Mochi, 2023-04-05:
> "check this box to ensure settings changes are preserved even after power cycling your Pinecil" but that's a bit long...

correct, that is what I mean.  you have longer detail like that show up  when  people toggle on "hints".
if  we already read the hint and have it memorized,  make the display words shorter. 

Currently  "Save changes to flash"  for any new user  doesn't tell us enough and we have to "Show hints"  the first time we use it anyway.  I had to read the hint and  I know a lot about how IronOS works compared to others. 

 once we have read the hint,  then a shorter form is good enough if my suggestion.

builder555, 2023-11-22:
It doesn't seem to be creating any issues for people, so I'll close this ticket for now. If I see others reporting similar issues, I'll reopen it and work on it.

closed 2023-11-22 by builder555