builder555/PineSAM #180: Cannot find a process with the process identifier xxxx

SabrinaRDC, 2023-04-16:
As the title said and shown in the picture below, for whatever reason the server.exe process doesn't appear/crashes and cannot be found by the script. I tried looking for the PID in something like Process Hacker 2 but couldn't find it manulally either. I tried this on two different Windows 11 machine but got the exact same problem every time. Remote execution policy is set to Signed on both machine.

I only know some very basics webdev stuff as a brief hobby and I don't know python so any help would be greatly appreciated. Thanks.

[image]

River-Mochi, 2023-04-16:
I tested this version on windows 11, and  it is working (except for some minor known issues but they don't prevent it from running). 

 more details needed.
 
1. please  follow the Template for new issue ticket so we have all the details and and they are listed in an easy manner to find in one place.

This is part of the template that is  most important for this kind of problem:

**Specs:**
 - OS: [e.g. Windows 11]
 - Browser: [e.g. chrome, safari]
 - PineSAM Version: [e.g. 1.0.8]
 - IronOS build/date: [e.g. 2.21.5096470, 09-04-23]

is 2.21 ironsOS installed and is Bluetooth enabled?
if the browser did not start at all,  which browser is default on your PC?
Even  if  bluetooth is "off" on the Pinecil, the server and the main_server  terminals will still start, it will just say in the terminal that  "device not found" 

Right now  it seems to be a problem on windows pc or the install method.   

3.  not sure what  Process Hacker 2 is,  but it should not be needed.    The PID can be found just using Windows task manager, all running PID will be in there.  if  powershells says the PID can not be found then it also will not be found in  Windows task manager or elsewhere. Most likely because it crashed and the pid went away.

4.  there should be other Terminals that open when start.ps1  script is started.  one for main_server and one for server. the screenshot cuts off the top of the pwsh window so can't see the version or if other tabs are opened.
There is a Terminal where the script is started and this  shows the Process ID's that are running for the other two terminals that open.  one is main_server.exe and the other  separate terminal is for server.exe
This image is of  what a working one looks like where  as long as those two server terminals are running, the main  powershell screen where I started  start.ps1  will  just hang there waiting. 
[image]

First thing to look at is why  windows might not be letting server.exe run?
also since not screenshot was provided, did main_server.exe run?

I added the process ID  or PID number just as a way  to show that something is happening, otherwise people would just see a blank hanging after starting "start.ps1". if you look at the script it will just  Wait and stay open as long as main_server or server.exe is running.

5.  not sure of a policy called  "Signed" 
please give screen shot and details of  which command  was typed in powershell to show the permissions and what is output of  Powershell permissions. see the Discussions section for how to get this. follow example in Discussions screenshot to get output and  give screenshot information. 

6. how was the original Pinesam  zip file extracted?  was Zip  properties checked to "Unblock" it or was it just extracted?
 - screen shot of the properties of PineSAM zip file would help. 
 - follow these additional screenshots and  hints  for extracting the Zip  and giving the permissions  from the Discussions tab: 
 https://github.com/builder555/PineSAM/discussions/106

7. which version of Powershell?  update to  newest  7.3.3 or 7.3.4?  and also update  windows 11 too so  we are on same page and starting point.  I currently tested PineSAM  to work on both  7.3.3.  and  7.3.4  powershell. 

8.   Which steps were followed for the  Windows premade binary version?
was it these steps in this link?  and if yes,
 https://builder555.github.io/PineSAM/install-guide/easy-install/#windows

then
- did you notice anything in the steps that did "not"  happen exactly like the steps? 
- for example, did you see  Windows Defender Pop-up  windows like see in  4. screenshots from this Discussions hints?
and  did you "Allow access"  accept  Both  Pop-up windows when they appeared?
-  screenshots for examples.  https://github.com/builder555/PineSAM/discussions/106#discussion-4960445
- sometimes it is minimized need look in task bar and Accept.
- check also  Windows  Defender> Advanced > inbound  rules  for server  and main_server  it should all be green.  if it was blocked it or rejected  to Allow it to run and see any  Red circles here, delete all red circles. 
all the main_server and  server rules should be green.  if anything is Red circle in here, it is being blocked. 
[image]
another way  is  delete all the green and all the red blocked circles  for  "main_server" and "server"   and then when  start.ps1 is started again the   Windows defender will pop-up again.  Then  allowed and accepted both of them.  Inside inbound rules, can right click main_server and server and then see the path they go to are inside the correct pinesam folder as well.

9.  For the Windows premade binary version
if you followed  all the exact instructions from here
(a)   https://builder555.github.io/PineSAM/install-guide/easy-install/#windows
and it did not work 
(b)  then  follow all the additional tips and steps from this post which also has detailed screen shots
https://github.com/builder555/PineSAM/discussions/106#discussion-4960445

10.  Recommend  deleting the zip  and  all the Inbound  rules  in Windows defender for   "main_server" and "server" entries  related to pinesam (right click them to check if they are in path to folder).   Reinstall  zip  following the additional  tips in the Discussions section linked.

===
11. if above doesn't work, then next step is perhaps  try the other method of install which is  

- Build from Scripts:  https://builder555.github.io/PineSAM/install-guide/dev-scripts/#windows-dev
-  Highly recommended to first  follow  instruction / hints  in this post for installing  Python and Nodjs, updating Powershell,  before doing the  Build from Scripts Windows Dev build instructions. https://github.com/builder555/PineSAM/discussions/7#discussion-4862766

River-Mochi, 2023-04-17:
I wrote a different Start  script for  powershell if you want to try it.  this checks for more things and gives more error messages.
put it in the same folder where you have   Start.ps1   and  unblock and extract this zip.  
execute it by typing  start02.ps1     ( I made it a different name so it will not over-write your existing start.ps1).

### [start02.zip](https://github.com/builder555/PineSAM/files/11256096/start02.zip)

builder555, 2023-11-22:
Without the feedback from the original reporter, can't fix the problem.

closed 2023-11-22 by builder555