builder555/PineSAM #181: Error handling message for windows Bluetooth

River-Mochi, 2023-04-16:
**Is your feature request related to a problem? Please describe.**
If bluetooth services in Windows is toggled "off" or there is no working bluetooth on the PC, then PineSAM script  will exit because the Process IDs are killed. 

This image simulates  having bluetooth toggled **off** on windows:
[image]

**Describe the solution you'd like**
1.  is it possible for PineSAM to check if windows has bluetooth running and let  user know that "Bluetooth services must be turned On"
2.  If  (1)  can not be done where PineSAM checks if windows bluetooth service is running, then  is it possible to add  extra  handling "hint".   I assume that "cannot find a process with the process identifier"  is a built-in system error message  from Get-Process command.  I could try  to improve the  script.ps1   so  if it sees  "cannot find a process with the process identifier"   that it then suggests  to  "HINT:  check windows bluetooth settings is on and functional"

Example of  how the Setting should look in windows when it's ON:

[image]

River-Mochi, 2023-04-17:
I wrote a new windows script that test  if  Bluetooth Toggle is ON or OFF. 
I need to find testers to make sure it works on windows 10 also.   Zip is attached.

works on my  windows 11 everytime if I toggle off or on the buttons.  This section is part that checks the Bluetooth toggle:
```
# Check Bluetooth status
$bluetoothAdapter = Get-NetAdapter | Where-Object { $_.InterfaceDescription -like "*Bluetooth*" }
if ($bluetoothAdapter) {
    Write-Host "Good: Windows Bluetooth is Enabled." -ForegroundColor Green
} else {
    Write-Host "== WINDOWS BLUETOOTH IS DISABLED. TOGGLE IT ON, AND TRY AGAIN ==" -ForegroundColor Red
	exit 1
}

```

[start02.zip](https://github.com/builder555/PineSAM/files/11256065/start02.zip)

River-Mochi, 2023-04-18:
This script has only been tested by  2x windows people so far.  I'm looking for more testers. So far it works.

1.  adds a few Error handlers
2. checks if bluetooth is on in windows settings - and tells user to Enable it.
3. checks if script is being executed from the pinesam folder otherwise path to main_server will fail - Tells user.
4. lets user know that Browser is starting. 
5. shows  PID  of both servers, and if one server fails, then Error message is printed to restart the terminal script.

### [start02.zip](https://github.com/builder555/PineSAM/files/11257033/start02.zip)

builder555, 2024-01-27:
the starter script has been removed, so this issue is no longer relevant.

closed 2024-01-27 by builder555