builder555/PineSAM #185: Can't run from iCloud Drive on macOS 13

joelcarlton, 2023-04-20:
**To Reproduce**
Steps to reproduce the behaviour:
Install dev version to a folder in iCloud drive and then run ./run-dev.sh

**Expected behaviour**
It should run without errors.

**Actual behaviour**
Python crashes

Termination Reason:    Namespace TCC, Code 0 
This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app's Info.plist must contain an NSBluetoothAlwaysUsageDescription key with a string value explaining to the user how the app uses this data.

**Screenshots**

**Specs:**
 - OS: macOS 13.3
 - Browser: Safari
 - PineSAM Version: 1.13.12
 - IronOS build/date: 2.21

**Additional context**

builder555, 2023-04-20:
did you check [Known Issues](https://github.com/builder555/PineSAM#-known-issues)?

joelcarlton, 2023-04-20:
I did but I missed the section on bluetooth. Thanks; I'll close.

```
bleak causes Python to crash on Mac: https://github.com/hbldh/bleak/issues/768

possible solution: give access to iTerm (or whichever terminal you use) to Bluetooth in Settings
```

closed 2023-04-20 by joelcarlton

River-Mochi, 2023-04-20:
> I did but I missed the section on bluetooth. Thanks; I'll close.

just so we know to help other people  what part of the bluetooth was issue?

joelcarlton, 2023-04-20:
I had to add iTerm to the list of applications allowed to access Bluetooth.

River-Mochi, 2023-04-20:
> I had to add iTerm to the list of applications allowed to access Bluetooth.

this is great, I could add some instructions to the web Usage guide for people trying to run from MacOS Cloud. 
- does this mean  people can install  this on mac iCloud  and then run it from any  browser?  It does not need to be on same home network?
- whatever details you can provide would be great since I don't have a mac.

joelcarlton, 2023-04-21:
> this is great, I could add some instructions to the web Usage guide for people trying to run from MacOS Cloud.
> 
> does this mean people can install this on mac iCloud and then run it from any browser? then device like your phone does not need to be on your same home network as the laptop/pc running backend pinesam server?
> 

On the Mac, iCloud Drive presents itself as a mounted volume/folder like Dropbox or Google Drive would. I cloned the repo to iCloud Drives's file space and ran the back/front end on the local Mac, the same as if I installed it somewhere else on the hard disk. This is NOT an iCloud App and does not allow iPhones/iPads to load the frontend over the internet. Mobile devices wanting to show the frontend web content would still need to either be on the LAN or use port forwarding or VPN.

> whatever steps you can provide would be great since I don't have a mac and this might help others.

I misdiagnosed the issue being related to iCloud Drive. The root issue was always related to iTerm not being given the proper access to Bluetooth. Sorry for the confusion.