builder555/PineSAM #189: [security] Backend server is world-accessible.

dreirund, 2023-07-25:
Ahoj (@builder555),

I think I have discovered a security issue:

While the UI by default listens only on localhost, and only with the `--host` parameter can be made to listen on outwards facing network interfaces, the backend server listens on all interfaces:

```
cd backend
python main_server.py
```

prints to the terminal

```
21:45:39.055::INFO::Starting pinecil monitor
21:45:39.055::INFO::waiting for pinecil...
21:45:39.056::INFO::Detecting "pinecil"...
21:45:39.142::INFO::server listening on 0.0.0.0:12999
[...]
```

and when I connect from another computer in my network to the computer running `main_server.py` to port 12999,  
`nc 192.168.100.4 12999`,  
and issue an HTTP GET request,  
`GET /`,  
I get answer:  
```
HTTP/1.1 400 Bad Request
Date: Tue, 25 Jul 2023 19:46:23 GMT
Server: Python/3.11 websockets/10.4
Content-Length: 77
Content-Type: text/plain
Connection: close

Failed to open a WebSocket connection: did not receive a valid HTTP request.
```

So when I now would use a WebSocket client I can connect to the backend server from anywhere in the internet, if my computer has a publicly facing IP address (which with IPv6 is quite often the case).

I would expect that per default it also listens only on localhost.

#### Specs:
 - OS: [Artix GNU/Linux](http://artixlinux.org/) (rolling release)
 - Browser: Firefox version 116.0b6
 - PineSAM Version: Latest git; `git describe --tags`: `v1.14.0-4-gcaf2ad3`.
 - IronOS version: 2.21

Regards!

builder555, 2023-07-25:
this was a design choice - to allow people using the UI from anywhere on their network.
generally it's a bad idea to expose your computer directly to the internet, and i don't know if listening to 0.0.0.0 would expose it on ipv6? i haven't tried.

I could make the host an env var and default to 127.0.0.1, but considering that several people had issues even starting this, i have a feeling i'll get many more complaints about it not being accessible over network when they open the UI on their phone.

dreirund, 2023-07-26:
> this was a design choice - to allow people using the UI from anywhere
> on their network. generally it's a bad idea to expose your computer
> directly to the internet,

but why then you by default limit the listening connection for the
webserver, when you assume "demilitarised zone"?

And I know people who exactly want this, for reasons.

Also local network can be malicious, and "stuff can happen", so
designing every layer secure is a good design principle in my opinion,
even if this layer normally should not exposed, it can prevent bad
stuff if for some security issue somewhere else it is.

> and i don't know if listening to 0.0.0.0
> would expose it on ipv6? i haven't tried.

I think you are right, but I don't know, I also have not dug into IPv6
until now.

> I could make the host an env var and default to 127.0.0.1, but
> considering that several people had issues even starting this, i have
> a feeling i'll get many more complaints about it not being accessible
> over network when they open the UI on their phone.

To make things less complicated (for the user), you can make the
backend and the GUI server all into one executable, then there is also
only one network connection to listen to, and easier to start up
manually.

Regards!

builder555, 2023-07-26:
> but why then you by default limit the listening connection for the webserver, when you assume "demilitarised zone"? 
I'm not limiting it either, it is listening to 0.0.0.0 by default as well:
`"dev": "vite --port 8080 --host 0.0.0.0"` (in ui/package.json)

> And I know people who exactly want this, for reasons. Also local network can be malicious, and "stuff can happen", so designing every layer secure is a good design principle in my opinion, even if [...]

true, and i'd have to either build it into a single executable (not listening to any connections), or restrict to localhost. in either case I'd lose the ability to use it from a phone or a tablet.

> To make things less complicated (for the user), you can make the backend and the GUI server all into one executable, then there is also only one network connection to listen to, and easier to start up manually. Regards!

how would I be able to access it from another device on the network? if i make it all into one executable, it would have to run entirely on the machine, not in a browser (most browsers don't have good support for BLE).

dreirund, 2023-07-26:
Ahoj,

> > but why then you by default limit the listening connection for the
> > webserver, when you assume "demilitarised zone"?   
> 
> I'm not limiting it either, it is listening to 0.0.0.0 by default as
> well: `"dev": "vite --port 8080 --host 0.0.0.0"` (in ui/package.json)

Ah OK, then it is the way I am starting it (trying to avoid wrappers
over wrappers I directly start `node_modules/vite/bin/vite.js` in the `ui/` dir), and it defaults to localhost.

OK, got it, I saw you make it different in the recommended way of running the thing!

> and i'd have to either build it into a single executable (not
> listening to any connections), or restrict to localhost. in either
> case I'd lose the ability to use it from a phone or a tablet.

Not necessarily if it can be specified via command line options.

> 
> > To make things less complicated (for the user), you can make the
> > backend and the GUI server all into one executable, then there is
> > also only one network connection to listen to, and easier to start
> > up manually. Regards!  
> 
> how would I be able to access it from another device on the network?

Still via the web browser!

Now you have the python backend server (`backend/main_server.py`) that listens for the node.js webserver (`ui/*`) that listens for the webbrowser.

If you combine the backend server and the webserver into one, you can still connecto to it via the webbrowser. And this single programme can receive a command line option telling it where/ how to listen.

builder555, 2024-01-25:
I combined the code into a single executable. now you can run it as `./Pinecil 127.0.0.1 8080` so it'll only accept connections from the same machine. 
By default it still listens to 0.0.0.0. Try it out: https://github.com/builder555/PineSAM/releases/tag/v2.1.14

dreirund, 2024-01-26:
Closing. I [don't use it anymore](https://aur.archlinux.org/packages/pinesam-git#comment-953461).

closed 2024-01-26 by builder555