builder555/PineSAM #190: Make the backend server exit gracefully on `<CTRL>`.

dreirund, 2023-07-25:
#### Is your feature request related to a problem? Please describe.

I am used to start stuff from the terminal.

Since the backend server has no (apparent) way to be gracefully terminated from within, I terminate it with `<CTRL>`.

It thereby emits error messages due to the abortion.

I know from other projects (but do not remember how) that in python it is possible to gracefully handle `<CTRL>+C` (which actually sends a `SIGINT` signal or `KeyboardInterrupt` to the process) and exit in an orderly manner.

#### Describe the solution you'd like.

Handle `SIGINT` or `KeyboardInterrupt` and exit in an orderly manner.

#### Additional context.

I start the backend server with `python backend/main_server.py`.

When I press `<CTRL>+C`, the following might be printed to the terminal (it differs from time to time, depending on which path the programme is currently in while receiving the signal):

```
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/[...]/.cache/yay/pinesam-git/src/pinesam/backend/main_server.py", line 138, in main
    await asyncio.gather(*tasks)
  File "/home/[...]/.cache/yay/pinesam-git/src/pinesam/backend/main_server.py", line 125, in pinecil_monitor
    pinecil_data = await pinecil.get_live_data()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/[...]/.cache/yay/pinesam-git/src/pinesam/backend/pinecil_ble.py", line 231, in get_live_data
    values = await self.__read_live_data(self.crx_bulk_data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/[...]/.cache/yay/pinesam-git/src/pinesam/backend/pinecil_ble.py", line 206, in __read_live_data
    raw_value = await self.ble.read_characteristic(crx)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/[...]/.cache/yay/pinesam-git/src/pinesam/backend/ble.py", line 66, in read_characteristic
    return await self.client.read_gatt_char(handle) #type: ignore
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/__init__.py", line 637, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 730, in read_gatt_char
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/dbus_fast/aio/message_bus.py", line 371, in call
    await future
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/[...]/.cache/yay/pinesam-git/src/pinesam/backend/main_server.py", line 142, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 123, in run
    raise KeyboardInterrupt()
KeyboardInterrupt
```

Another example:

```
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/[...]/.cache/yay/pinesam-git/src/pinesam/backend/main_server.py", line 138, in main
    await asyncio.gather(*tasks)
  File "/home/[...]/.cache/yay/pinesam-git/src/pinesam/backend/main_server.py", line 114, in pinecil_monitor
    await pinecil.connect()
  File "/home/[...]/.cache/yay/pinesam-git/src/pinesam/backend/pinecil_ble.py", line 105, in connect
    await self.ble.ensure_connected()
  File "/home/[...]/.cache/yay/pinesam-git/src/pinesam/backend/ble.py", line 31, in ensure_connected
    await self.__detect_device_address()
  File "/home/[...]/.cache/yay/pinesam-git/src/pinesam/backend/ble.py", line 40, in __detect_device_address
    devices = await BleakScanner.discover()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/__init__.py", line 243, in discover
    await asyncio.sleep(timeout)
  File "/usr/lib/python3.11/asyncio/tasks.py", line 639, in sleep
    return await future
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/[...]/.cache/yay/pinesam-git/src/pinesam/backend/main_server.py", line 142, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 123, in run
    raise KeyboardInterrupt()
KeyboardInterrupt
```

Regards!

builder555, 2023-07-25:
thanks. I've been meaning to get to it, i remember the asyncio was giving me issues when trying to capture sigint. i'll take a look this weekned.

builder555, 2024-01-27:
Solved in [v2.1.15](https://github.com/builder555/PineSAM/releases/tag/v2.1.15)

closed 2024-01-27 by builder555