builder555/PineSAM #191: Arch Linux package created.

dreirund, 2023-07-25:
I just want to leave a not that I have created an Arch Linux package out of this:

[&nearr;&hairsp;`pinesam-git`](https://aur.archlinux.org/packages/pinesam-git).

It was a bit of a nightmare, since nodejs is so f\*\*\*\*\*g broken when it comes to proper operating system packaging ... *(a simple terminal interface with optional gtk or qt GUI would have been so much simpler ;-) -- but that is how it is, I live with it)*

builder555, 2023-07-25:
wow that's awesome! i haven't used arch in years, maybe this'll give me a chance to try again.

i like vue.js too much to write gui in GTK. but a terminal interface sounds like an interesting idea. maybe in Norton Commander style?.

dreirund, 2023-07-26:
On Tue, 25 Jul 2023 16:23:11 -0700, builder555
***@***.***> wrote:

> i like vue.js too much to write gui in GTK. but a terminal interface
> sounds like an interesting idea. maybe in Norton Commander style?.

That would mean (n)curses.

I was thinking of a line-oriented thing like "normal" text-absed
programmes: One execution -- one information or, depending on
given arguments, option(s) set.

But a full-screen self-updating tool like `htop` could also be
interesting.

Regards!

leumasme, 2023-11-17:
N

leumasme, 2023-11-17:
No need to keep the frontend dependencies around after bundling the Vue app. If the frontend is already built and served via the python server, there's no need to have any nodejs or npm dependencies around at all.  
If the bluetooth communication is then moved to the frontend via webbluetooth, although this would drop support for firefox since they refuse to implement it, it would make the whole project statically servable so everyone can use the same instance (or bundle it all into a single .html file to open locally if you prefer).

builder555, 2023-11-24:
> No need to keep the frontend dependencies around after bundling the Vue app. If the frontend is already built and served via the python server, there's no need to have any nodejs or npm dependencies around at all. If the bluetooth communication is then moved to the frontend via webbluetooth, although this would drop support for firefox since they refuse to implement it, it would make the whole project statically servable so everyone can use the same instance (or bundle it all into a single .html file to open locally if you prefer), but this is probably not feasible due to compatability.

There is a project that is pure browser-based: https://github.com/joric/pinecil

The issue is that it doesn't work on iOS browsers (at least the more popular ones). Until BLE is natively and properly supported in all browsers, I'll have to stick with a backend.

builder555, 2023-11-24:
added a reference to this package in readme.

closed 2023-11-24 by builder555