builder555/PineSAM #193: Fails to work if version.txt is not available in main server location

andypiper, 2023-11-22:
**To Reproduce**
Steps to reproduce the behaviour:
1. Download current (1.14.1) package for macOS
2. Execute `./start.sh` and navigate to page in browser
3. Switch on Pinecil

**Expected behaviour**
PineSAM should start to display data from the Pinecil in the browser window

**Actual behaviour**
The browser never completely loads the UI and data is never shown.
An error appears in the console:

```text
13:24:40.602::INFO::server listening on 0.0.0.0:12999
13:24:45.605::WARNING::Device not found
13:24:46.608::INFO::waiting for pinecil...
13:24:46.608::INFO::Detecting "pinecil"...
127.0.0.1 - - [22/Nov/2023 13:24:48] "HEAD / HTTP/1.1" 200 -
127.0.0.1 - - [22/Nov/2023 13:24:49] "GET / HTTP/1.1" 304 -
13:24:49.779::INFO::connection open
13:24:49.869::INFO::Got message: {"command":"GET_APP_INFO"}
13:24:49.870::ERROR::connection handler failed
Traceback (most recent call last):
  File "websockets/legacy/server.py", line 240, in handler
  File "main_server.py", line 92, in ws_handler
  File "main_server.py", line 74, in handle_message
  File "main_server.py", line 42, in process_command
  File "main_server.py", line 22, in read_app_version
FileNotFoundError: [Errno 2] No such file or directory: '/Users/andypiper/Downloads/PineSAM-MacOS-x64-v1.14.1/main_server/version.txt'
13:24:49.885::INFO::connection closed
```

**Screenshots**
[image]

**Specs:**
 - OS: macOS Sonoma 14.1.1
 - Browser: Chrome
 - PineSAM Version: 1.14.1
 - IronOS build/date: 2.22

**Additional context**
This issue is trivially solved by adding a `version.txt` file to the `main_server` directory:

```shell
$ echo "1.14.1" > main_server/version.txt
```

Once that is in place, the UI loading completes and the iron can be controlled and viewed.

Potentially there are a couple of solutions:

- ensure `version.txt` is added to the packaged builds correctly
- don't block the UI loading if the version cannot be displayed underneath the GH logo/link (but still issue a console warning)

Hope this helps! Thanks for the nice work on the software 👍🏻

builder555, 2023-11-22:
confirmed! thanks for the report.

builder555, 2023-11-22:
please take a look, it should be closed now. feel free to reopen

closed 2023-11-22 by builder555

andypiper, 2023-11-23:
Thanks! Confirming that the new package runs right out of the box.