builder555/PineSAM #199: Move code related to interacting with the PINECIL to separate library for use by other projects

DanieleQ97, 2023-12-29:
**Is your feature request related to a problem? Please describe.**
The code used for talking to the PINECIL is at the moment embedded in the backend part of PineSAM (in the files [ble.py](https://github.com/builder555/PineSAM/blob/master/backend/ble.py), [pinecil_ble.py](https://github.com/builder555/PineSAM/blob/master/backend/pinecil_ble.py), [crx_uuid_name_map.py](https://github.com/builder555/PineSAM/blob/master/backend/crx_uuid_name_map.py) and [pinecil_setting_limits.py](https://github.com/builder555/PineSAM/blob/master/backend/pinecil_setting_limits.py)).
If someone wanted to make another python application for interacting with the PINECIL (I, for one, would like to write one using [Textual](https://github.com/Textualize/textual) to avoid having to use the browser and do everything in the terminal), they would need to:
- Reinvent the wheel and write all the code from scratch.
- Fork PineSAM, for what is basically an entirely different application except for part of the backend.
- Copy the needed files from this project to a new one and put disclaimers everywhere "🚨 This code was copied from PineSAM! 🚨".

All of which seem less than ideal solutions. Even for PineSAM itself because it would make contributing upstream from the derived project back to that part of PineSAM a pain.

**Describe the solution you'd like**
Move the relevant part of the project to a separate library. This would not only enable other developers to reuse your code for different projects but also likely simplify maintenance and improvement efforts for PineSAM.

**Additional context**
No additional context.

builder555, 2023-12-29:
better modularity sounds like a great idea. 
do you have any suggestions for the interface of this library?

DanieleQ97, 2023-12-30:
I was giving a look at the code and, at least for a "version zero", I think that keeping the files mostly as they are could work quite well (perhaps just renaming some parts here and there).
The init file of the library could export the Pinecil class and the four exceptions (two from ble.py and two from pinecil_ble.py).

This should also require minimal changes to [main_server.py](https://github.com/builder555/PineSAM/blob/master/backend/main_server.py) (in theory just the imports, which would become simpler).
From:
```python
from ble import DeviceNotFoundException
from ble import DeviceDisconnectedException
from pinecil_ble import Pinecil
from pinecil_ble import InvalidSettingException
from pinecil_ble import ValueOutOfRangeException
```
To something like:
```python
from <NEW_LIBRARY> import Pinecil
from <NEW_LIBRARY> import (
                            DeviceNotFoundException,
                            DeviceDisconnectedException,
                            InvalidSettingException,
                            ValueOutOfRangeException
                          )
```

The relevant tests should also be moved: [test_pinecil.py](https://github.com/builder555/PineSAM/blob/master/backend/test_pinecil.py), [test_data.py](https://github.com/builder555/PineSAM/blob/master/backend/test_data.py) and a copy of [test_utils.py](https://github.com/builder555/PineSAM/blob/master/backend/test_utils.py)

Does this looks like a good approach to you?

builder555, 2023-12-30:
i think i get the idea, i'll get started, then post a link here to the new repo

builder555, 2023-12-31:
started the library here https://github.com/builder555/pinecil_lib

DanieleQ97, 2023-12-31:
Awesome, thank you!
I'll check it out ASAP to see if I can help somehow.

builder555, 2024-01-01:
It's basically done, can be used in any project. I just can't publish right now since PyPi admins are on vacations.

builder555, 2024-01-02:
now you can use `pip install pinecil`: https://pypi.org/project/pinecil

closed 2024-01-02 by builder555

DanieleQ97, 2024-01-02:
There is some issue or todo from PineSAM that might be worth moving to and addressing in the new library?
Maybe [Show BLE MAC address](https://github.com/builder555/PineSAM/issues/121), [Integrate Check Authenticity inside PineSAM](https://github.com/builder555/PineSAM/issues/83) and [Make BLE pairing process more transparent](https://github.com/builder555/PineSAM/issues/81)

builder555, 2024-01-02:
The first two require firmware changes, the last one isn't applicable here, since you would be the one implementing the entire pairing process (e.g. searching for devices, connecting to it, getting the settings, updating settings). The library only provides the functions, but doesn't have any pairing process on its own.