builder555/PineSAM #203: ModuleNotFoundError: No module named 'pinecil'

slabua, 2024-01-14:
**To Reproduce**
Steps to reproduce the behaviour:
1. `$ pip install pinecil`
2. `$ ./start.sh`

**Expected behaviour**
The webserver is started and the PineSAM can be used normally

**Actual behaviour**
```
Traceback (most recent call last):
  File "main_server.py", line 8, in <module>
ModuleNotFoundError: No module named 'pinecil'
[2288200] Failed to execute script 'main_server' due to unhandled exception!
Opening in existing browser session.
```

**Screenshots**
[image]

**Specs:**
 - OS: Fedora 39
 - Browser: Chrome 121.0.6167.57 (Official Build) beta (64-bit)
 - PineSAM Version: 1.14.4
 - IronOS build/date: 2.22E.C308FE8C, 05-11-23]

builder555, 2024-01-14:
Can you try the latest release? https://github.com/builder555/PineSAM/releases/tag/v1.14.5 I think I missed a dependency when building a package.

slabua, 2024-01-14:
@builder555 thanks for checking it so quickly
this time I got another missing dependency
```
04:08:31.263::INFO::Starting pinecil monitor
04:08:31.264::INFO::Detecting "pinecil"...
Traceback (most recent call last):
  File "main_server.py", line 152, in <module>
  File "asyncio/runners.py", line 194, in run
  File "asyncio/runners.py", line 118, in run
  File "asyncio/base_events.py", line 684, in run_until_complete
  File "main_server.py", line 148, in main
  File "main_server.py", line 116, in pinecil_monitor
  File "pinecil/pinecil.py", line 334, in find_pinecils
  File "pinecil/ble.py", line 20, in find_device_addresses
  File "bleak/__init__.py", line 317, in discover
  File "bleak/__init__.py", line 145, in __init__
  File "bleak/backends/scanner.py", line 284, in get_platform_scanner_backend_type
  File "PyInstaller/loader/pyimod02_importers.py", line 419, in exec_module
  File "bleak/backends/bluezdbus/scanner.py", line 5, in <module>
  File "PyInstaller/loader/pyimod02_importers.py", line 419, in exec_module
  File "dbus_fast/__init__.py", line 1, in <module>
  File "PyInstaller/loader/pyimod02_importers.py", line 419, in exec_module
  File "dbus_fast/introspection.py", line 5, in <module>
  File "PyInstaller/loader/pyimod02_importers.py", line 419, in exec_module
  File "dbus_fast/errors.py", line 57, in <module>
  File "src/dbus_fast/message.py", line 1, in init dbus_fast.message
ModuleNotFoundError: No module named 'dbus_fast._private.marshaller'
[118413] Failed to execute script 'main_server' due to unhandled exception!
Opening in existing browser session.
```

builder555, 2024-01-14:
hm, that's new. let me look into it

builder555, 2024-01-15:
i might've fixed it, give the latest one a try: https://github.com/builder555/PineSAM/releases/tag/v1.14.6

slabua, 2024-01-15:
@builder555 thanks! It's working now~

closed 2024-01-15 by builder555