builder555/PineSAM #217: [CRITICAL] Tapping tempture change repeatedly causes sudden jumps and uncontrolled temps

parkerlreed, 2024-09-20:
**To Reproduce**
Steps to reproduce the behaviour:
1. Turn on heater on Pinecil
2. Use Pinecil interface to adjust temp clicking repeatedly

**Expected behaviour**
Temperature to only jump by the amount of clicks registered.

**Actual behaviour**
Temperatures rises uncontrollably.

**Screenshots**

https://github.com/user-attachments/assets/91043d0f-21fc-4807-b0a3-994e85870260

**Specs:**
 - OS: Arch Linux
 - Browser: Firefox/Chrome
 - PineSAM Version: 2.2.6
 - IronOS build/date: 2.22 11/05/2023

**Additional context**

Clicking slowly works as expected, but if you tap repeatedly in either direction, the temps continues without user intervention until it starts trying to set invalid values. Was at a low temp and just wanted to go up a few degrees and ended up at max without intervention. Serious safety issue.

Tested with both Firefox and Chrome

parkerlreed, 2024-09-20:
https://github.com/user-attachments/assets/749dd3b5-cb01-4a85-a4b0-8cec6505823a

If it goes on long enough the iron itself starts flashing the display like it's glitching out.

builder555, 2024-09-20:
Sounds like you found a hidden "don't tire your mouse out, I got this" feature :)
Thank you for the report, I'll take a look this weekend.

builder555, 2024-09-21:
@parkerlreed do you want to try 2.2.9? I just pushed it. I didn't install arch, but I might've found the problem. If it persists, I'll set up a vm and test it out. 
Which desktop environment are you using?

parkerlreed, 2024-09-22:
Of course I forgot to bring it with me on the trip.

Plasma Wayland

I'll update whenever I can test it.

parkerlreed, 2024-11-08:
Yep all looks good