builder555/PineSAM #220: Handle temp is not changing to F scale, only shows as C scale.

LucidEye, 2024-12-28:
**To Reproduce**
Steps to reproduce the behaviour:
1. Set Pinecil v2 temp scale to F in 'User Interface' settings.
2. Open PineSAM app and connect Pinecil v2 with BLE.
3. All temps displayed in app are in F scale except for the handle temp which still shows as C scale.

**Expected behaviour**
All temps in app display should be in the same scale as what is selected in the Pinecil firmware settings.

**Actual behaviour**
Handle temp display in PineSAM app does not change to F scale when firmware on Pinecil is set to F even though all other temps are displayed in correct scale.

**Screenshots**
[image]

**Specs:**
 - OS: Ubuntu MATE Linux v22.04.5 LTS,  
 - DE: MATE desktop environment v1.26.0
 - Browser: Chrome
 - PineSAM Version: v2.2g
 - IronOS build/date: v2.22E.C308FE8C 05-11-23

builder555, 2025-01-04:
Oops it closed automatically, I'll reopen it, give this a try, and if it's fixed, feel free to close the issue: 
https://github.com/builder555/PineSAM/releases/tag/v2.2.12

LucidEye, 2025-01-05:
Well now I can't get the app to start at all :-(
My Pinecil is connected, paired, and trusted.
I have Chrome browser open.
I downloaded the Ubuntu version, unzipped it to a folder, and created a menu icon for it and when I click the icon nothing happens
Both files have permission to run as an executable.
Tried manually entering http://localhost:8080/ in the browser and it can't find the page.
Tried running it manually in terminal and it throws this error...

[PYI-42070:ERROR] Failed to load Python shared library '/tmp/_MEILiVJ4o/libpython3.11.so.1.0': dlopen: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by /tmp/_MEILiVJ4o/libpython3.11.so.1.0)

I have Python 3.10.12 installed

Ran apt update and apt full-upgrade and still get the same error.

parkerlreed, 2025-01-05:
It's compiled for too new of a glibc. Your distro has an older version.

Even with compatible glibc, it crashes.

```
[parker@parker-framework PineSAM-Ubuntu-x64-v2.2.12]$ ./PineSAM 
01:39:39 INFO     01:39:39.913::INFO::Starting websocket server on             ws_server.py:137
                  0.0.0.0:8080                                                                 
         INFO     01:39:39.916::INFO::Starting pinecil monitor            pinecil_monitor.py:40
         INFO     01:39:39.917::INFO::looking for pinecil...              pinecil_monitor.py:43
         INFO     01:39:39.917::INFO::looking for pinecil...              pinecil_monitor.py:21
         INFO     01:39:39.918::INFO::Detecting "pinecil"...                          ble.py:20
Traceback (most recent call last):
  File "main.py", line 75, in <module>
  File "asyncio/runners.py", line 190, in run
  File "asyncio/runners.py", line 118, in run
  File "asyncio/base_events.py", line 654, in run_until_complete
  File "main.py", line 64, in main
  File "ws_server.py", line 138, in serve
  File "websockets/asyncio/server.py", line 802, in __init__
TypeError: BaseEventLoop.create_server() got an unexpected keyword argument 'create_protocol'
[PYI-96039:ERROR] Failed to execute script 'main' due to unhandled exception!
```

LucidEye, 2025-01-05:
So how can I fix this?  
I tried installing Python 3.12 and it is still throwing the same error.
Can I just install the library it's looking for?
If so how?

parkerlreed, 2025-01-05:
I'm saying there's two issues here. Your issues is incompatible glibc. This can't be fixed without recompiling the project on your system.

Second is if it works the program has a bug and doesn't run.

LucidEye, 2025-01-05:
Unfortunately I don't have much experience with manually compiling anything so I would need concise instructions on how to do that.
But you say it would still fail to work... so I'm assuming that's something the developer would need to fix.
In the meantime I can just go back to the previous version I was using.

commit 433653fd (2025-01-04): fix: handle temp always in C (fixes #220)