builder555/PineSAM #223: Issue when launching PINESAM.exe on Windows 11

Rinkakazi, 2025-01-11:
Hi, code noob here and wanted to ask what this error means when launching PINESAME.exe

```
02:32:03 INFO     02:32:03.599::INFO::Starting websocket server on 0.0.0.0:8080                                                                                                            ws_server.py:137
         INFO     02:32:03.604::INFO::Starting pinecil monitor                                                                                                                        pinecil_monitor.py:40
         INFO     02:32:03.606::INFO::looking for pinecil...                                                                                                                          pinecil_monitor.py:43
         INFO     02:32:03.607::INFO::looking for pinecil...                                                                                                                          pinecil_monitor.py:21
         INFO     02:32:03.608::INFO::Detecting "pinecil"...                                                                                                                                      ble.py:20
Traceback (most recent call last):
  File "main.py", line 75, in <module>
  File "asyncio\runners.py", line 190, in run
  File "asyncio\runners.py", line 118, in run
  File "asyncio\base_events.py", line 654, in run_until_complete
  File "main.py", line 64, in main
  File "ws_server.py", line 138, in serve
  File "websockets\asyncio\server.py", line 802, in __init__
TypeError: BaseEventLoop.create_server() got an unexpected keyword argument 'create_protocol'
[PYI-35412:ERROR] Failed to execute script 'main' due to unhandled exception!
```

builder555, 2025-01-16:
Unfortunately Windows is always a pain for me to test. I'll try to get a hold of a windows machine to troubleshoot the issue when I can.

Quba1, 2025-01-21:
I see the same error on Ubuntu 24.04

Quba1, 2025-01-21:
After a bit of testing (on Linux) I think I have identified the cause of this issue. `websockets` 14.0 [changed their `asyncio` implementation](https://websockets.readthedocs.io/en/stable/project/changelog.html#id7) and [deprecated the `create_protocol` arg](https://websockets.readthedocs.io/en/stable/howto/upgrade.html#create-protocol-create-connection).

I tested a few releases and it seems that regression was introduced in `v2.2.12`. So possibly GH Actions script started using a newer version of `websockets`.

There seems to be a few solutions:
- Building with `websockets < 14.0`, I'm not familiar with `pyinstaller` so I don't know how that can be done;
- Replacing `from websockets import serve` with `from websockets.legacy.server import serve`, which seems to be the easiest;
- Updating the code to use new `websockets` implementation as described in [upgrade guide](https://websockets.readthedocs.io/en/stable/howto/upgrade.html), which obviously is the most difficult option.

I'm happy to help with rectifying this issue

builder555, 2025-01-21:
Thank you for your PR @Quba1 ! I merged it, the new build is available, give it a try: https://github.com/builder555/PineSAM/releases/tag/v2.2.14

I had to add a couple of dummy commits, because I forgot how my own CD pipeline works 🤦

closed 2025-02-01 by builder555